me-no-dev/ESPAsyncWebServer #534: What can I do with this ? [W][AsyncTCP.cpp:902] _poll(): rx timeout 4

GeorgeFlorian, 2019-06-26:
Hello !

Following the new updates and commits made by @me-no-dev I've updated all of my projects to include the newest version of this library.

I can see some improvements. 
What I can also see is the following warning `[W][AsyncTCP.cpp:902] _poll(): rx timeout 4` whenever the server slows down.

I don't really understand what it does, but it seems it keeps the server from crashing.

Can I get some explanation regarding the warning ? Thank you !

me-no-dev, 2019-06-26:
I have found that some computers/browsers seem to cause LwIP to emit a new connect event, but the connection is never actually completed. The warning you get is a result of the AsyncTCP timing out waiting this connection to happen and is clearing it up.

closed 2019-06-26 by me-no-dev

GeorgeFlorian, 2019-06-26:
> I have found that some computers/browsers seem to cause LwIP to emit a new connect event, but the connection is never actually completed. The warning you get is a result of the AsyncTCP timing out waiting this connection to happen and is clearing it up.

Great ! Thanks !

GeorgeFlorian, 2019-06-26:
What about: `[W][AsyncTCP.cpp:882] _poll(): pcb is NULL` ?

me-no-dev, 2019-06-26:
it is there temporarily to show places where the heap would have been corrupted before

GeorgeFlorian, 2019-06-26:
> it is there temporarily to show places where the heap would have been corrupted before

Ok. Great ! I get a lot of those. And in return, the server isn't crashing.

me-no-dev, 2019-06-26:
that is the point :) I will remove the debug later on, but for now it's a good indication of what is going on.

virat92, 2019-11-14:
i am getting same warning [W][AsyncTCP.cpp:913] _poll(): rx timeout 4 
after getting this warnings my Webserver Disconnect with browser especially when i update my firmware from browser(chrome ). and i have to restart my ESP manually.
can you please tell me the solution