me-no-dev/ESPAsyncWebServer #536: Is there a fix to "This site can’t be reached" ?

GeorgeFlorian, 2019-07-01:
Hello !

So I am using an ESP32 EVB to control two relays over the WebServer and it sometimes freezes without crashing the ESP and thus making the ESP unusable.

I am using a WDT, just as @me-no-dev advised me to, but it doesn't make any difference. The Server still remains frozen.

```
void setup(){
  enableCore1WDT();
  delay(500);
  Serial.begin(115200);
  delay(1000);
// ...
      server.on("/relay1/on", HTTP_GET, [](AsyncWebServerRequest *request){
        // request->send(200, "text/plain",relay1 + " is ON");
        digitalWrite(RELAY1, HIGH);
        status1 = "ON";
        logOutput((String)relay1 + " is ON");
        request->redirect("/home");
      });

      server.on("/relay1/off", HTTP_GET, [](AsyncWebServerRequest *request){
        // request->send(200, "text/plain", relay1 + " is OFF");
        digitalWrite(RELAY1, LOW);
        status1 = "OFF";
        logOutput((String)relay1 + " is OFF");
        request->redirect("/home");
      });

      server.on("/relay1_on", HTTP_POST, [](AsyncWebServerRequest * request){
        int params = request->params();
        String values_button = "";
        for(int i=0;i<params;i++){
          AsyncWebParameter* p = request->getParam(i);
          if(p->isPost()){
              logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
              values_button = p->name();            
            } else {
                logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
              }
        } // for(int i=0;i<params;i++)
        values_button.trim();
        if(values_button == "relay1_on") {
          request->redirect("/relay1/on");
          // relayFlagOne = true;
          startTimeRelayOne = millis();
          Serial.println("Do I get here ? relay1_on POST");
        }   
      });
      server.on("/relay1_off", HTTP_POST, [](AsyncWebServerRequest * request){
        int params = request->params();
        String values_button = "";
        for(int i=0;i<params;i++){
          AsyncWebParameter* p = request->getParam(i);
          if(p->isPost()){
              logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
              values_button = p->name();            
            } else {
                logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
              }
        } // for(int i=0;i<params;i++)
        values_button.trim();
        if(values_button == "relay1_off") {
          request->redirect("/relay1/off");
        }  
      });
```
```
            <div class = "placeholders">
                <span>%PLACEHOLDER_Relay1%</span>
                <form method="post" action= "/relay1_on">
                    <input class="button" name="relay1_on" type="submit" value="/on">
                </form>
                <form method="post" action= "/relay1_off">
                    <input class="button" name="relay1_off" type="submit" value="/off">
                </form>
            </div>
```
The only feedback I got was this:
```
POST[relay1_on]: /on

Do I get here ? relay1_on POST
Relay1 is ON
POST[relay2_on]: /on

Do I get here ? relay2_on POST
Relay2 is ON
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:730] _poll(): ack timeout 4
[W][AsyncTCP.cpp:730] _poll(): ack timeout 4
[W][AsyncTCP.cpp:730] _poll(): ack timeout 4
Closing Relay1 in: 10seconds !
Relay1 closed
Closing Relay2 in: 10seconds !
Relay2 closed
```

And then nothing. The web-site was unreachable. I had to hit the reset button.

Is there a way of this not happening ?

vinot, 2019-07-15:
Hi,

did not try, sorry.

but just to confirm some steps; did you ping it? afaik esp32 uses power saves that can be disabled 
 [https://esp32.com/viewtopic.php?t=5537](url)

aslo with wdt I supose,,,

the hard way: ethereal

GeorgeFlorian, 2019-07-16:
> Hi,
> 
> did not try, sorry.
> 
> but just to confirm some steps; did you ping it? afaik esp32 uses power saves that can be disabled
> [https://esp32.com/viewtopic.php?t=5537](url)
> 
> aslo with wdt I supose,,,
> 
> the hard way: ethereal

I did not ping ! I will try pinging it whenever I get the chance.
Topic: https://esp32.com/viewtopic.php?t=5537 does not exist.

Also, I have a `enableCore1WDT();` at the beginning of the `void setup()` and it does not help me.

I haven't tried Ethernet and I do not care about Ethernet right now ! I need Wi-Fi and that is it.

vinot, 2019-07-16:
Hi,

sorry for the link, that's right: https://github.com/espressif/arduino-esp32/issues/1484

Wireless ping goes to 5-7ms in my simple network (before it was +200ms)

GeorgeFlorian, 2019-07-18:
> Hi,
> 
> sorry for the link, that's right: [espressif/arduino-esp32#1484](https://github.com/espressif/arduino-esp32/issues/1484)
> 
> Wireless ping goes to 5-7ms in my simple network (before it was +200ms)

`Esp_wifi_set_ps (WIFI_PS_NONE);` looks like it works. I will continue testing it.

closed 2019-10-05 by stale[bot]