me-no-dev/ESPAsyncWebServer #538: How do I download files from SPIFSS ?

GeorgeFlorian, 2019-07-03:
Hello !

As the tittle says, I would like to know if it's possible to download files from the SPIFFS file system.

Pablo2048, 2019-07-03:
Did you look at the examples? (https://github.com/me-no-dev/ESPAsyncWebServer/tree/master/examples/ESP_AsyncFSBrowser)

GeorgeFlorian, 2019-07-03:
> Did you look at the examples? (https://github.com/me-no-dev/ESPAsyncWebServer/tree/master/examples/ESP_AsyncFSBrowser)

No, I haven't looked at them in a loong while. I knew there was a FSBrowser example but I had no idea it contained a download method.

And I am looking right now at it and I don't understand much.
I've only used `AsyncWebServer server(80);` until now and I've just `send.on()` my whole server.

I don't know what `AsyncWebSocket ws("/ws");` or `AsyncEventSource events("/events");` mean or why would I need them.

And there is also ArduinoOTA. ` //Send OTA events to the browser`. What does this mean ?

Also, the word `download` is nowhere to be found in there. I am soo lost...

Pablo2048, 2019-07-03:
So give it a try... ArduinoOTA is - as it says, support for OTA firmware update driectly from IDE/VSC, websocket is... well just web socket need for bidirectional communication between browser and server. Word ```download``` doesn't need to be in the source code, but in the web application source (data subdirectory). Websockets and events are standard web technlogies certainly described somewhere (RFC?) and also here https://github.com/me-no-dev/ESPAsyncWebServer#async-websocket-plugin - please don't waste my time asking for such things...

GeorgeFlorian, 2019-07-03:
> So give it a try... ArduinoOTA is - as it says, support for OTA firmware update driectly from IDE/VSC, websocket is... well just web socket need for bidirectional communication between browser and [...]

C'mon, be a pal. It's not like there's a course on "How to use ESPAsyncWebServer". 
Also, you are pretty knowledgeable, but I've just started working with the library .
I do not have a Networking background. It's the first time in my life I hear WebSocket and WebSocket Event.
I am sorry if these look like stupid questions, but it's hard for a newbie to understand what's going on.

closed 2019-07-03 by GeorgeFlorian

Pablo2048, 2019-07-03:
Oh, come on - why do you think, that I have networking background? I've got my knowledge by reading, searching and thinking... If you try this example, then you maybe got the picture within 5 minutes - way shorter time than me trying to write explanations with my bad english :-) ...
Ok - so FSBrowser is two stage thing - a backend (async web server) and a frontend (html/js appliaction, running in your browser). Both parts need to communicate with other one so here the websocket jumps in. Events are small messages from server (backend) to the browser (frontend) and here are used for OTA progress notification. Overall what this example does is that it allows you to manipulate the SPIFFS content (upload, download, browse, modify and delete SPIFFS content).

GeorgeFlorian, 2019-07-03:
So I did what you told me and uploaded the code and the data into the ESP32.
All I have is a black window with the following messages on it.
```
Events Opened
Event: hello!
Connected
TXT:Hello Client 8 :)
TXT:I got your text message
$
```
And next to the $ there's an input.

I don't even know how to use this damn thing.

Pablo2048, 2019-07-03:
Did you upload the spiffs (aka /data directory) to the ESP32?

GeorgeFlorian, 2019-07-03:
> Did you upload the spiffs (aka /data directory) to the ESP32?

Yes, I did.

Pablo2048, 2019-07-03:
Ok, so just try to enter ```http://<esp3 ip address>/edit```

GeorgeFlorian, 2019-07-03:
> Ok, so just try to enter `http://<esp3 ip address>/edit`

Ok. That worked ! Thank you !

So it was this: `server.addHandler(new SPIFFSEditor(SPIFFS, http_username,http_password));`. That has a handler on `/edit`. Great.