me-no-dev/ESPAsyncWebServer #540: Does serveStatic() refreshes the web-page automatically ?

GeorgeFlorian, 2019-07-03:
Hello !

I'm trying to replace all of the handlers I have: `server.on()` with `serveStatic()`, because as I understand it, I can serve/GET all the file with just one handler and then `setDefaultFile()`. But I will still need to make `server.on()` if I want to POST.

I started a new project and gave it a try:
```
myWebServer.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html").setCacheControl("max-age=0");
myWebServer.begin();
```

I've also tried it without `setCacheControl("max-age=0");`, but it changed nothing.

And I've noticed that this handler refreshes the page automatically every few seconds.
Also, on each refresh I get:
`[W][AsyncTCP.cpp:903] _poll(): rx timeout 4` or `[W][AsyncTCP.cpp:896] _poll(): ack timeout 4`

And the Warnings keep piling up:
```
[W][AsyncTCP.cpp:883] _poll(): pcb is NULL
[W][AsyncTCP.cpp:903] _poll(): rx timeout 4
[W][AsyncTCP.cpp:903] _poll(): rx timeout 4
[W][AsyncTCP.cpp:903] _poll(): rx timeout 4
[W][AsyncTCP.cpp:896] _poll(): ack timeout 4
[W][AsyncTCP.cpp:896] _poll(): ack timeout 4
[W][AsyncTCP.cpp:896] _poll(): ack timeout 4
[W][AsyncTCP.cpp:896] _poll(): ack timeout 4
[W][AsyncTCP.cpp:896] _poll(): ack timeout 4
```

closed 2019-07-03 by GeorgeFlorian

atanisoft, 2019-07-03:
refresh is controlled by your html code or the browser side. The server side will not initiate a refresh unless you tell it to.

GeorgeFlorian, 2019-07-03:
> refresh is controlled by your html code or the browser side. The server side will not initiate a refresh unless you tell it to.

Yes ! I figured this out seconds after posting here.
I have used an old `.html` that I found on my PC and it has `<meta http-equiv="refresh" content="5"/>`.