me-no-dev/ESPAsyncWebServer #542: How do you upload files ?

GeorgeFlorian, 2019-07-04:
Hello !

So I am trying to upload files to the ESP32 using this server and a browser and I can't figure it out...

```

GeorgeFlorian, 2019-07-04:
I have tried another handler:
```
      server.on("/upload", HTTP_POST, [](AsyncWebServerRequest *request) {
        if(request->hasParam("data", true, true) && SPIFFS.exists(request->getParam("data", true, true)->value()))
          request->send(200, "", "UPLOADED: "+request->getParam("data", true, true)->value());
      });
```

But this returns `not found`.

atanisoft, 2019-07-04:
The approach using
```
 server.on("/upload", HTTP_POST, [](AsyncWebServerRequest *request) {
        request->send(200);
      }, handleUpload);
```

Is how to handle this. The handleUpload method will be called as the data arrives so it can be processed. This is covered in the examples and in the [readme](https://github.com/me-no-dev/ESPAsyncWebServer#file-upload-handling).

There is no equivalent to ```HTTPUpload& uploadfile = server.upload()``` nor a real need for it.

GeorgeFlorian, 2019-07-04:
> The approach using
> 
> ```
>  server.on("/upload", HTTP_POST, [](AsyncWebServerRequest *request) {
>         request->send(200);
>       }, handleUpload);
> ```
> 
> Is how to handle this. The handleUpload method will be called as the data arrives so it can be processed. This is covered in the examples and in the [readme](https://github.com/me-no-dev/ESPAsyncWebServer#file-upload-handling).
> 
> There is no equivalent to `HTTPUpload& uploadfile = server.upload()` nor a real need for it. What are you trying to do, OTA?

I am trying to upload `.txt` files. By doing this I want to give to the user the possibility to Import data. That data will be parsed and used by the rest of the code in the project.

So this is the handler from the readme and is the same I am using.
```
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
  if(!index){
    Serial.printf("UploadStart: %s\n", filename.c_str());
  }
  for(size_t i=0; i<len; i++){
    Serial.write(data[i]);
  }
  if(final){
    Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index+len);
  }
}
```

But it doesn't work.

atanisoft, 2019-07-04:
Very likely there is something else missing in your code then. This is a very basic http task that is known working.

GeorgeFlorian, 2019-07-04:
I've changed the handlers like this:
```
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
  if(!index){
    logOutput((String)"UploadStart: " + filename);
  }
  for(size_t i=0; i<len; i++){
    Serial.write(data[i]);
  }
  if(final){
    logOutput((String)"UploadEnd: " + filename + "," + index+len);
    request->redirect("/files");
  }
}
      server.on("/doUpload", HTTP_POST, [](AsyncWebServerRequest *request) {},
      [](AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data,
                    size_t len, bool final) {handleUpload(request, filename, index, data, len, final);}
      );
```

And the form is in a html file called: `files.html`.

```
      server.on("/files", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(SPIFFS, "/files.html", "text/html");
      });
```

And here is what's inside `files.html`:
```
<div class = "upload">
	<form method = "POST" action = "/doUpload" enctype="multipart/form-data">
		<input type="file" name="data"/>
		<input type="submit" name="upload" value="Upload" title = "Upload Files">
	</form>	
</div>
```

atanisoft, 2019-07-04:
And can you verify how the browser is sending the upload? Nothing appears to be incorrect on the esp code which you have provided. The double wrapper for the /doUpload upload handler is not needed though, you can just pass handleUpload to the server.on method as the last arg.

GeorgeFlorian, 2019-07-04:
Ok. Sooo, writing the handler like that made the following command work: `Serial.write(data[i]);`. But the file isn't actually uploaded.

GeorgeFlorian, 2019-07-04:
I will now try it like this:
```
      server.on("/doUpload", HTTP_POST, [](AsyncWebServerRequest *request) {
      }, handleUpload);
```
And also check what the browser does.

atanisoft, 2019-07-04:
> made the following command work: `Serial.write(data[i]);`. But the file isn't actually uploaded.

If Serial.write(...) is being invoked then the file **IS** being uploaded but you aren't doing anything with the data as it is being sent to that method.

GeorgeFlorian, 2019-07-04:
> > made the following command work: `Serial.write(data[i]);`. But the file isn't actually uploaded.
> 
> If Serial.write(...) is being invoked then the file **IS** being uploaded but you aren't doing anything with the data as it is being sent to that method.

Well... yes... I thought that `handleUpload` will upload to SPIFFS. But it doesn't seems like that.
I need it to write to SPIFFS.
I think that I need to replace: 
```
  for(size_t i=0; i<len; i++){
    Serial.write(data[i]);
  }
```
with something like:
``
File file = SPIFFS.open(filename,"w");
for(size_t i=0; i<len; i++) {
file.print(i);
}
file.close();
``` 

Does that look fine to you ?

atanisoft, 2019-07-04:
> Well... yes... I thought that `handleUpload` will upload to SPIFFS. But it doesn't seems like that.

Why would it write to SPIFFS directly? The purpose of the callback is for *YOUR* code to do something meaningful with the incoming data stream. 

> with something like:
> 
> ```
> File file = SPIFFS.open(filename,"w");
> for(size_t i=0; i<len; i++) {
> file.write(data[i]);
> }
> file.close();
> ```
> 
> Does that look fine to you ?

Not really. You will only record part of the data and then truncate the file for the next chunk. You should also use ```file.write(data, len)``` to send the entire block to the filesystem in one call.

Refer to https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L528-L544 for how to properly handle the streaming of data to the filesystem.

GeorgeFlorian, 2019-07-04:
> Why would it write to SPIFFS directly? The purpose of the callback is for _YOUR_ code to do something meaningful with the incoming data stream.
>
>
> Not really. You will only record part of the data [...]

So like this:
```
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
  if(!index){
    logOutput((String)"UploadStart: " + filename);
  }
  File file = SPIFFS.open(filename, "w");
  if(len) {
    file.write(data,len);
  }
  
  if(final){
    logOutput((String)"UploadEnd: " + filename + "," + index+len);
    file.close();
    // request->redirect("/files");
    request->send(200, "text/plain", "File Uploaded !");
  }
}
```

atanisoft, 2019-07-04:
close but not quite right, you will still truncate the file with your updated method. Try this:
```
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
  if(!index){
    logOutput((String)"UploadStart: " + filename);
    // open the file on first call and store the file handle in the request object
    request->_tempFile = SPIFFS.open(filename, "w");
  }
  if(len) {
    // stream the incoming chunk to the opened file
    request->_tempFile.write(data,len);
  }
  if(final){
    logOutput((String)"UploadEnd: " + filename + "," + index+len);
    // close the file handle as the upload is now done
    request->_tempFile.close();
    request->send(200, "text/plain", "File Uploaded !");
  }
}
```

GeorgeFlorian, 2019-07-04:
What is the difference between: 
`request->_tempFile = SPIFFS.open(filename, "w");` and `File file = SPIFFS.open(filename, "w");`
and
`request->_tempFile.write(data,len);` and `file.write(data,len);`
?

GeorgeFlorian, 2019-07-04:
Also, it doesn't work like this. It sends me a "notFound" error.

atanisoft, 2019-07-04:
It allows you to open the file once and append to it as the data comes in. With your locally scoped approach you will end up truncating and overwriting the file with every chunk.

> Also, it doesn't work like this. It sends me a "notFound" error.

Then there is a problem with how the browser is sending the data to the server *OR* you have not registered the endpoint correctly on the server.

The SPIFFSEditor uses this approach, I use this approach, many others use this exact approach without issues.

GeorgeFlorian, 2019-07-04:
I get that it should work. But changing from local scope to `request->_tempFile` made it not work.

And the other handlers are pretty simple:
```
      server.on("/files", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(SPIFFS, "/files.html", "text/html");
      });

      server.on("/files", HTTP_POST, handle_config);

      server.on("/doUpload", HTTP_POST, [](AsyncWebServerRequest *request) {
      }, handleUpload);
```

```
void handle_config(AsyncWebServerRequest * request) {
  Serial.print("handle_config, request: "); 
  Serial.println(request->methodToString());
  
  // listHeaders(request);
  // listParams(request);

  if (request->hasParam("filename", true)) { // Download
    if (request->hasArg("download")) { // file download
      Serial.println("Download Filename: " + request->arg("filename"));
      AsyncWebServerResponse *response = request->beginResponse(SPIFFS, request->arg("filename"), String(), true);
      response->addHeader("Server", "ESP Async Web Server");
      request->send(response);
      return;
    }
  }

  String HTML; // HTML code 
  String filename = request->url() + ".html";
  File pageFile = SPIFFS.open(filename, "r");
  if (pageFile) {
    HTML = readString(pageFile);
    pageFile.close();
    HTML = addDirList(HTML);
    request->send(200, "text/html", HTML);
  }
}
```

So I have a handler for `/files` and a handler for `/doUpload` that is a form inside `/files`.
The handler is basically making a table with what ever `.txt` files it founds and also gives the possibility of downloading the respective files.

Can those request get mixed up ?

GeorgeFlorian, 2019-07-04:
Update. The only time I get errors is whenever I'm trying to upload `.txt` files. Everything else works. So that means that it must be the other handler that is at fault here. The `handle_config` from `server.on("/files", HTTP_POST, handle_config);`

atanisoft, 2019-07-04:
> I get that it should work. But changing from local scope to `request->_tempFile` made it not work.

local scope will only work if the file is very small and will fit fully in one chunk (usually very small). If moving to request->_tempFile broke it then it was already broken to begin with as this is how SPIFFSEditor works and it doesn't have issues like you are seeing.

> Update. The only time I get errors is whenever I'm trying to upload `.txt` files. Everything else works. So that means that it must be the other handler that is at fault here. The `handle_config` from `server.on("/files", HTTP_POST, handle_config);`

Sounds like your html code is not consistently calling your /doUpload handler, you could combine /doUpload and /files POST together and make it simpler :)

GeorgeFlorian, 2019-07-04:
> Sounds like your html code is not consistently calling your /doUpload handler, you could combine /doUpload and /files POST together and make it simpler :)

I could, couldn't I ? Too bad I have no idea how.
Make one handler for the whole page ?
But what would I write for the `server.on` ?

How do one combines these:
```
      server.on("/files", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(SPIFFS, "/files.html", "text/html");
      });

      server.on("/files", HTTP_POST, handle_config);

      server.on("/doUpload", HTTP_POST, [](AsyncWebServerRequest *request) {
      }, handleUpload);
```
into one handler ?

I have made a lot of handlers like this. I have pages with 2-3 buttons and each button has his own handler.
I had no idea you can combine them. I can't even imagine it

Each form has his own handler, right ?
And if that ain't right, how can I combine the following:
## index.html
```
<form method="POST" action="/index">
   <input type="text" />
   <input type="submit" />
</form>
<form method="POST" action="/button1">
   <input type="submit" name="go_to_page2" />
</form>
<form method="POST" action="/button2">
   <input type="submit" name="go_to_page3" />
</form>
```
And on the server side I would've made handlers for each of it:
```
server.on("/index", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", "text/html");
});
server.on("/index", HTTP_POST, handler_that_saves_input_values);
server.on("/button1", HTTP_POST, handler_that_redirects_to_page1);
server.on("/button2", HTTP_POST, handler_that_redirects_to_page2);
```

And now you are telling me that I can do that with one handler ?  MIND literately BLOWN !

atanisoft, 2019-07-04:
try this:
```
server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");
server.on("/", HTTP_POST,
  [](AsyncWebServerRequest *request) {
    if(request->hasArg("go_to_page2")) {
// button 1
      request->redirect("/page2");
    } else if(request->hasArg("go_to_page3")) {
// button 2
      request->redirect("/page3");
    } else {
// handle default case here
    }
  });
server.onFileUpload(handleUpload);
```
With this approach any GET request to the server will try and find a file in SPIFFS and serve it to the client, if they do not request a specific file it will send index.html. For any POST request it checks for the name of the submit button. For any upload to *ANY* page it will call the handleUpload method.

GeorgeFlorian, 2019-07-04:
> try this:
>
> ```
> server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");
> server.on("/", HTTP_POST,
>   [](AsyncWebServerRequest *request) {
>     if(request->hasArg("go_to_page2")) {
> // button [...]

Wow.. this looks way tidier than what I currently have on my server:

Here, for shits and giggles:
```
 server.on("/register", HTTP_GET, [](AsyncWebServerRequest *request){              
    request->send(SPIFFS, "/indexAP.html", "text/html");
  });
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){              
    request->redirect("/register");
  });
  server.on("/register", HTTP_POST, [](AsyncWebServerRequest * request){
    int params = request->params();
    String values_user[2];
    for(int i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isPost()){
          logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
          values_user[i] = p->value();            
        } else {
            logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
          }
    } // for(int i=0;i<params;i++)
    
    if(values_user[0] != NULL && values_user[0].length() > 4 &&
      values_user[1] != NULL && values_user[1].length() > 7) {
          File userWrite = SPIFFS.open("/user.txt", "w");
          if(!userWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write USER credentials !");
          userWrite.println(values_user[0]);  // Username
          userWrite.println(values_user[1]);  // Password
          userWrite.close();
          logOutput("Username and password saved !");          
          request->redirect("/IP-Config");
          // digitalWrite(LED, LOW);
    } else request->redirect("/register");  
  }); // server.on("/register", HTTP_POST, [](AsyncWebServerRequest * request)

  server.on("/button", HTTP_POST, [](AsyncWebServerRequest * request){
    int params = request->params();
    String values_button = "";
    for(int i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isPost()){
          logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
          values_button = p->value();            
        } else {
            logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
          }
    } // for(int i=0;i<params;i++)
    values_button.trim();
    if(values_button == "Skip") {
      request->redirect("/IP-Config");
    }   
  });

  server.on("/IP-Config", HTTP_GET, [](AsyncWebServerRequest *request){      
    request->send(SPIFFS, "/AP_configPage.html", "text/html");
  });
  server.on("/logs", HTTP_GET, [](AsyncWebServerRequest* request){      
    request->send(SPIFFS, "/events.html", "text/html");
  });
  server.on("/events_placeholder.html", HTTP_GET, [](AsyncWebServerRequest* request){      
[... 74 lines cut ...]
        } else {
            logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
          }
    }
    if(values_dhcp[0] != NULL && values_dhcp[0].length() != 0 &&
      values_dhcp[1] != NULL && values_dhcp[1].length() != 0) {
      File inputsWrite = SPIFFS.open("/inputs.txt", "w");
      if(!inputsWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write DHCP IP credentials !");
      inputsWrite.println(values_dhcp[0]);  // SSID
      inputsWrite.println(values_dhcp[1]);  // Password
      inputsWrite.close();
      logOutput("Configuration saved !");
      logOutput("Restarting in 5 seconds...");
      request->redirect("/logs");
      restartFlag = true;
    } else request->redirect("/dhcpIP");
  
  }); // server.on("/dhcpLogin", HTTP_POST, [](AsyncWebServerRequest * request)
  
  server.onNotFound([](AsyncWebServerRequest *request){
    request->redirect("/register");
  });

  server.begin(); //-------------------------------------------------------------- server.begin()
```

GeorgeFlorian, 2019-07-05:
> try this:
>
> ```
> server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");
> server.on("/", HTTP_POST,
>   [](AsyncWebServerRequest *request) {
>     if(request->hasArg("go_to_page2")) {
> // button [...]

One quick question !
I have multiple pages on the server. And I serve them like this:
```
  server.on("/dhcpIP", HTTP_GET, [](AsyncWebServerRequest *request){      
    request->send(SPIFFS, "/dhcpIP.html", "text/html");
  });
```
So the server knows that `/dhcpIP` is `/dhcpIP.html`.

But, using the following I don't know where to write `/dhcpIP` and where to write `/dhcpIP.html`.
```
server.serveStatic("/", SPIFFS, "/").setDefaultFile("indexAP.html").setTemplateProcessor(processor);
server.on("/", HTTP_POST,[](AsyncWebServerRequest *request) { 
});
```
**How will the server know which one is `/dhcpIP` ?**

In the following example I have 2 forms on 2 different pages with the same action: `/dhcpIP`.

This is what I've written but it's confusing:
Page1:
```
        <form method = "post" action="/dhcpIP">
            <input class = "button" type="submit" name="dhcp" value="DHCP IP" title="Redirect to DHCP IP Config page">
            <input class = "button" type="submit" name="static" value="Static IP" title="Redirect to Static IP Config page" formaction="/staticIP">
        </form>
```
DHCP IP Page
```
        <form method="POST" action="/dhcpIP">
            <div class ="text_input">            
                <input type="text" placeholder="Network Name (SSID)" name="networkNameDHCP" value="" pattern=".{5,30}" required title="Enter between 5 and 30 characters">
            </div>        
            <div class="text_input">            
                <input type="password" placeholder="Password" name="networkPasswordDHCP" value="" pattern=".{8,63}" required title="Enter between 8 and 63 characters">
            </div>        
            <input class="button" type="submit" name="saveDHCP" value="Save and Reboot">
        </form>
```

```
    } else if(request->hasArg("dhcp")) { // Pressed on DHCP Button from page 1
      request->redirect("/dhcpIP"); // Go To DHCP IP Page
    } else if(request->hasArg("saveDHCP")) { // pressed on SaveDHCP from DHCP IP Page
        int params = request->params();
        String values_dhcp[2];
        for(int i=0;i<params;i++){
          AsyncWebParameter* p = request->getParam(i);
          if(p->isPost()){
            logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
            values_dhcp[i] = p->value();            
          }
        } //for
        if(values_dhcp[0] != NULL && values_dhcp[0].length() != 0 &&
        values_dhcp[1] != NULL && values_dhcp[1].length() != 0) {
        File inputsWrite = SPIFFS.open("/network.txt", "w");
        if(!inputsWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write DHCP IP credentials !");
          inputsWrite.println(values_dhcp[0]);  // SSID
          inputsWrite.println(values_dhcp[1]);  // Password
          inputsWrite.close();
          logOutput("Configuration saved !");
          request->redirect("/logs");
          shouldReboot = true;
        } else request->redirect("/dhcpIP");
    }
```

GeorgeFlorian, 2019-07-05:
Or, for the above example I have `request->redirect("/logs");`. But `/logs` is actually `events.html`.
Using the old way I did it like this:
```
  server.on("/logs", HTTP_GET, [](AsyncWebServerRequest* request){
    request->send(SPIFFS, "/events.html", "text/html");
  });
```
But using the new way I have no idea how this should work.
The server will have no idea what is `/logs`...

GeorgeFlorian, 2019-07-05:
And it doesn't work. The CSS is not loading. The handlers don't work. And I have no other page other than `setDefaultFile("indexAP.html")`.
It also triggers a WTD which makes the ESP32 restart.

atanisoft, 2019-07-05:
@GeorgeFlorian I'd suggest taking a look at server.rewrite() which will map a uri (/logs) to another uri (/events.html).

GeorgeFlorian, 2019-07-05:
@atanisoft 

I'm gonna try that.

Meanwhile, I'm having trouble adding a processor to a send method.
```
String HTML;
request->send(200, "text/html", HTML);
```

And I can't find a way of adding a processor to that particular `send`.

I have done this before, but it uses another overload of `send`:
```
  server.on("/events_placeholder.html", HTTP_GET, [](AsyncWebServerRequest* request){
    request->send(SPIFFS, "/events_placeholder.html", "text/html", false, processor);
  });
```

atanisoft, 2019-07-05:
> I have done this before, but it uses another overload of `send`:
> 
> ```
>   server.on("/events_placeholder.html", HTTP_GET, [](AsyncWebServerRequest* request){
>     request->send(SPIFFS, "/events_placeholder.html", "text/html", false, processor);
>   });
> ```

This can be done with the serveStatic().setTemplateProcessor(proc) approach which I think you already found.

> Meanwhile, I'm having trouble adding a processor to a send method.
> 
> ```
> String HTML;
> request->send(200, "text/html", HTML);
> ```
> 
> And I can't find a way of adding a processor to that particular `send`.

Since HTML is expected to be a pre-formatted block it doesn't support templates by itself. You can add it though by using https://github.com/me-no-dev/ESPAsyncWebServer#chunked-response-containing-templates and using something like "HTML.substring(index, maxLen)" in the chunked callback code.

GeorgeFlorian, 2019-07-05:
> This can be done with the serveStatic().setTemplateProcessor(proc) approach which I think you already found.

Yes, I did found it, but that requires replacing my whole code and that will take a long time. I am currently trying to make it work the hard way: without using `serveStatic()`.

>Since HTML is expected to be a pre-formatted block it doesn't support templates by itself. You can add it though by using https://github.com/me-no-dev/ESPAsyncWebServer#chunked-response-containing-templates and using something like "HTML.substring(index, maxLen)" in the chunked callback code.

I'm gonna try that for now, until I make my project work as I want it, then I will try upgrading to `serveStatic()`.

GeorgeFlorian, 2019-07-05:
It's the only handler for `/files`.

```
  server.on("/files", HTTP_ANY, [](AsyncWebServerRequest *request){
    Serial.print("handle_config, request: ");
    Serial.println(request->methodToString());    

    if (request->hasParam("filename", true)) { // Download
      if (request->hasArg("download")) { // file download
        Serial.println("Download Filename: " + request->arg("filename"));
        AsyncWebServerResponse *response = request->beginResponse(SPIFFS, request->arg("filename"), String(), true);
        response->addHeader("Server", "ESP Async Web Server");
        request->send(response);
        return;
      }
    } else if(request->hasArg("goBack")) {
      request->redirect("register");
    }

    String HTML; // HTML code 
    String filename = request->url() + ".html";
    File pageFile = SPIFFS.open(filename, "r");
    if (pageFile) {
      HTML = readString(pageFile);
      pageFile.close();
      HTML = addDirList(HTML);
      // request->send(200, "text/html", HTML);
      AsyncWebServerResponse *response = 
      request->beginChunkedResponse("text/plain", [](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
        return HTML.substring(index, maxLen);
      }, processor);
      response->addHeader("Server","ESP Async Web Server");
      request->send(response);
    }
  });
```

And it gives me this: `an enclosing-function local variable cannot be referenced in a lambda body unless it is in the capture list`.

atanisoft, 2019-07-05:
```
AsyncWebServerResponse *response = 
      request->beginChunkedResponse("text/plain", [&](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
        return HTML.substring(index, maxLen);
      }, processor);
```

Maybe?

GeorgeFlorian, 2019-07-05:
> ```
> AsyncWebServerResponse *response = 
>       request->beginChunkedResponse("text/plain", [&](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
>         return HTML.substring(index, maxLen);
>       }, processor);
> ```
> 
> Maybe?

New error: `cannot convert 'String' to 'size_t {aka unsigned int}' in return`.

atanisoft, 2019-07-05:
```
AsyncWebServerResponse *response = 
      request->beginChunkedResponse("text/plain", [&](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
  String chunk = HTML.substring(index, maxLen);
  strcpy(buffer, chunk.c_str());
  return chunk.length();
}, processor);
```

You need to pass the chunk into the provided buffer and return the number of bytes put into the buffer.

GeorgeFlorian, 2019-07-05:
> ```
> AsyncWebServerResponse *response =
>       request->beginChunkedResponse("text/plain", [&](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
>   String chunk = HTML.substring(index, [...]

Buuuut... `buffer` is a `uint8_t` pointer and `chunk` will be a `char` pointer.
And it returns this error:
```
error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'char*' [-fpermissive]
```

atanisoft, 2019-07-05:
```
strcpy((char*)buffer, chunk.c_str());
```

MkLHX, 2019-07-05:
> 
> 
> Hello !
> 
> So I am trying to upload files to the ESP32 using this server and a browser and I can't figure it out...
> 
> I got this handler
> 
> ```
> void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
>   if(!index){
>     logOutput((String)"UploadStart: " + filename);
>   }
>   for(size_t i=0; i<len; i++){
>     Serial.write(data[i]);
>   }
>   if(final){
>     logOutput((String)"UploadEnd: " + filename + "," + index+len);
>   }
> }
> ```
> 
> And this handler inside setup():
> 
> ```
>       server.on("/upload", HTTP_POST, [](AsyncWebServerRequest *request) {
>         request->send(200);
>       }, handleUpload);
> ```
> 
> And this is in HTML:
> 
> ```
> 	<form method = "POST" action = /upload>
> 		<input type="file" name="data" data-multiple-caption="{count} files selected" multiple/>
> 		<input type="submit" name="upload" value="Upload" title = "Upload Files">
> 	</form>	
> ```
> 
> I know I am doing something wrong, but I have no idea what. I couldn't find anything useful in the examples or on Google. Maybe because I don't know what to search for.
> 
> Any ideas ?

hi there!

Trying exactly the same thing but no success .

Nothing wrote on Serial, and nothing happen when upload file on esp32.

My sketch:
``` 
#include "Arduino.h"
#include "WiFi.h"
#include "SPIFFS.h"
#include "ArduinoOTA.h"
#include "ESPAsyncWebServer.h"

const char *ssid = "<myssid>";
const char *password = "<mypwd>";

AsyncWebServer server(80);

void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final)
{
	if (!index)
	{
		Serial.printf("UploadStart: %s\n", filename.c_str());
	}
	for (size_t i = 0; i < len; i++)
	{
		Serial.write(data[i]);
	}
	if (final)
	{
		Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index + len);
	}
}
void setup()
{
	Serial.begin(115200);
	Serial.println("Booting");
	WiFi.mode(WIFI_AP_STA);
	WiFi.begin(ssid, password);
	while (WiFi.waitForConnectResult() != WL_CONNECTED)
	{
		Serial.println("Connection Failed! Rebooting...");
		delay(5000);
		ESP.restart();
	}
	ArduinoOTA
		.onStart([]() {
			String type;
			if (ArduinoOTA.getCommand() == U_FLASH)
				type = "sketch";
			else // U_SPIFFS
				type = "filesystem";

			// NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
			Serial.println("Start updating " + type);
		})
		.onEnd([]() {
			Serial.println("\nEnd");
		})
		.onProgress([](unsigned int progress, unsigned int total) {
			Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
		})
		.onError([](ota_error_t error) {
			Serial.printf("Error[%u]: ", error);
			if (error == OTA_AUTH_ERROR)
				Serial.println("Auth Failed");
			else if (error == OTA_BEGIN_ERROR)
				Serial.println("Begin Failed");
			else if (error == OTA_CONNECT_ERROR)
				Serial.println("Connect Failed");
			else if (error == OTA_RECEIVE_ERROR)
				Serial.println("Receive Failed");
			else if (error == OTA_END_ERROR)
				Serial.println("End Failed");
		});
	ArduinoOTA.begin();

	if (!SPIFFS.begin(true))
	{
		Serial.println(F("[main]... An Error has occurred while mounting SPIFFS"));
		return;
	}

	if (SPIFFS.exists("/logs.txt"))
	{
		SPIFFS.remove("/logs.txt");
	}
	File file = SPIFFS.open("/logs.txt", FILE_WRITE);
	file.clearWriteError();
	file.println("start writing logs");
	file.close();

	server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
		request->send(200, "application/json", "{\"status_message\":\"is alive\"}");
	});
	server.on("/upload-file", HTTP_GET, [](AsyncWebServerRequest *request) {
		String html = "<body><div><form method='post' action='/upload-file'><input type='file'><button>Send</button></form></div></body>";
		request->send(200, "text/html", html);
	});

	server.on("/upload-file", HTTP_POST, [](AsyncWebServerRequest *request) {
		AsyncWebServerResponse *response = request->beginResponse(200, "text/html", "hello world");
		response->addHeader("Connection", "close");
		request->send(response);
	},handleUpload);

	server.on("/download-file", HTTP_GET, [](AsyncWebServerRequest *request) {
		AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/logs.txt", String(), true);
		response->addHeader("Server", "ESP server");
		request->send(response);
	});
	server.onFileUpload(handleUpload);
	server.begin();

	Serial.println("Ready");
	Serial.print("IP address: ");
	Serial.println(WiFi.localIP());
}

void loop()
{
	ArduinoOTA.handle();
}
```

my goal is to success sending log.txt on esp32 spiffs through http post request.
in future i'd like to send firmware.bin in esp32 spiffs to permit OTA for an other ESP32 who connected on his AP network.

So @ this time, OTA on STA not work on ArduinoIDE and PlatformIO. On AP mode idem not success.

so read many issues, examples, discuss and still broken my head on this idea...

I'm crazy to thing it's possible?

GeorgeFlorian, 2019-07-07:
@MkLHX try this handler:
```
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
  if(!index){
    logOutput((String)"UploadStart: " + filename);
    // open the file on first call and store the file handle in the request object
    request->_tempFile = SPIFFS.open("/"+filename, "w");
  }
  if(len) {
    // stream the incoming chunk to the opened file
    request->_tempFile.write(data,len);
  }
  if(final){
    logOutput((String)"UploadEnd: " + filename + ",size: " + index+len);
    // close the file handle as the upload is now done
    request->_tempFile.close();
    request->redirect("/files");
  }
}
```

And call it inside setup using this: `server.onFileUpload(handleUpload);`

closed 2019-07-07 by GeorgeFlorian

GeorgeFlorian, 2019-07-07:
@atanisoft I also fixed the issue that I had with attaching the processor to that specific handler. I made it `PROGMEM` : 
```
  server.on("/files", HTTP_ANY, [](AsyncWebServerRequest *request){
    Serial.print("/files, request: ");
    Serial.println(request->methodToString());    

    if (request->hasParam("filename", true)) { // Download file
      if (request->hasArg("download")) { // file download
        Serial.println("Download Filename: " + request->arg("filename"));
        AsyncWebServerResponse *response = request->beginResponse(SPIFFS, request->arg("filename"), String(), true);
        response->addHeader("Server", "ESP Async Web Server");
        request->send(response);
        return;
      } else if(request->hasArg("delete")) { // Delete file
        SPIFFS.remove(request->getParam("filename", true)->value());
        // request->send(200, "", "DELETE: "+request->getParam("path", true)->value());
        request->redirect("/files");
      }
    } else if(request->hasArg("goBack")) { // GO Back Button
      request->redirect("register");
    }

    String HTML PROGMEM; // HTML code 
    String filename = request->url() + ".html";
    File pageFile = SPIFFS.open(filename, "r");
    if (pageFile) {
      HTML = readString(pageFile);
      pageFile.close();
      HTML = addDirList(HTML);
      AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", HTML.c_str(), processor);
      response->addHeader("Server","ESP Async Web Server");
      request->send(response);
    }
  });
```

I've also added the option to delete files.

Thank you for all your patience. I couldn't have done without your support.

MkLHX, 2019-07-09:
> 
> So @ this time, OTA on STA not work on ArduinoIDE and PlatformIO. On AP mode idem not success.
> 
Find part of solution (for BasicOTA) here https://github.com/espressif/arduino-esp32/issues/2965#issuecomment-509644324

So now i'm going to test your fileHandler thx

OscarHanzely, 2020-01-12:
It case it helps anybody, I implemented basic OTA based on source: https://lastminuteengineers.com/esp32-ota-web-updater-arduino-ide/
But I am using async web server so I used the hint provided in this thread. I don't chunk the file manually or flush to SPIFFS. Instead I use Update.h library to do all hard work. Code is pretty simple:

*OTA.h* 
this is basically the form from the source mentioned, but I removed that unnecessary login, since it does not protect anything anyways.
```

/* Style */
String style =
"<style>#file-input,input{width:100%;height:44px;border-radius:4px;margin:10px auto;font-size:15px}"
"input{background:#f1f1f1;border:0;padding:0 15px}body{background:#3498db;font-family:sans-serif;font-size:14px;color:#777}"
"#file-input{padding:0;border:1px solid #ddd;line-height:44px;text-align:left;display:block;cursor:pointer}"
"#bar,#prgbar{background-color:#f1f1f1;border-radius:10px}#bar{background-color:#3498db;width:0%;height:10px}"
"form{background:#fff;max-width:258px;margin:75px auto;padding:30px;border-radius:5px;text-align:center}"
".btn{background:#3498db;color:#fff;cursor:pointer}</style>";

/* Index page */
String indexPage = 
"<h1 style=\"color:white; text-align:center;\">ESP32 Garage Management</h1>" + style;

/* Firmware Page */
String firmwarePage = 
"<script src='https://ajax.googleapis.com/ajax/libs/jquery/3.2.1/jquery.min.js'></script>"
"<form method='POST' action='#' enctype='multipart/form-data' id='upload_form'>"
"<input type='file' name='update' id='file' onchange='sub(this)' style=display:none>"
"<label id='file-input' for='file'>   Choose file...</label>"
"<input type='submit' class=btn value='Update'>"
"<br><br>"
"<div id='prg'></div>"
"<br><div id='prgbar'><div id='bar'></div></div><br></form>"
"<script>"
"function sub(obj){"
"var fileName = obj.value.split('\\\\');"
"document.getElementById('file-input').innerHTML = '   '+ fileName[fileName.length-1];"
"};"
"$('form').submit(function(e){"
"e.preventDefault();"
"var form = $('#upload_form')[0];"
"var data = new FormData(form);"
"$.ajax({"
"url: '/update',"
"type: 'POST',"
"data: data,"
"contentType: false,"
"processData:false,"
"xhr: function() {"
"var xhr = new window.XMLHttpRequest();"
"xhr.upload.addEventListener('progress', function(evt) {"
"if (evt.lengthComputable) {"
"var per = evt.loaded / evt.total;"
"$('#prg').html('progress: ' + Math.round(per*100) + '%');"
"$('#bar').css('width',Math.round(per*100) + '%');"
"}"
"}, false);"
"return xhr;"
"},"
"success:function(d, s) {"
"console.log('success!') "
"},"
"error: function (a, b, c) {"
"}"
"});"
"});"
"</script>" + style;
```

*server.cpp*
```
#include <ESPAsyncWebServer.h>
#include <OTA.h>
#include <Update.h>

AsyncWebServer server(80);

void notFoundResponse(AsyncWebServerRequest *request)
{
    request->send(404, "text/plain", "Not found");
}

// handle the upload of the firmware
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final)
{
    // handle upload and update
    if (!index)
    {
        Serial.printf("Update: %s\n", filename.c_str());
        if (!Update.begin(UPDATE_SIZE_UNKNOWN))
        { //start with max available size
            Update.printError(Serial);
        }
    }

    /* flashing firmware to ESP*/
    if (len)
    {
        Update.write(data, len);
    }

    if (final)
    {
        if (Update.end(true))
        { //true to set the size to the current progress
            Serial.printf("Update Success: %ub written\nRebooting...\n", index+len);
        }
        else
        {
            Update.printError(Serial);
        }
    }
    // alternative approach
    // https://github.com/me-no-dev/ESPAsyncWebServer/issues/542#issuecomment-508489206
}

// mains erver handling block
void startServer()
{
    // web interface for OTA
    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        AsyncWebServerResponse *response = request->beginResponse(200, "text/html", indexPage);
        response->addHeader("Connection", "close");
        request->send(response);
    });
    server.on("/firmware", HTTP_GET, [](AsyncWebServerRequest *request) {
        AsyncWebServerResponse *response = request->beginResponse(200, "text/html", firmwarePage);
        response->addHeader("Connection", "close");
        request->send(response);
    });
    // handling uploading firmware file
    server.on("/update", HTTP_POST, [](AsyncWebServerRequest *request) {
        if (!Update.hasError()) {
            AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "OK");
            response->addHeader("Connection", "close");
            request->send(response);
            ESP.restart();
        } else {
            AsyncWebServerResponse *response = request->beginResponse(500, "text/plain", "ERROR");
            response->addHeader("Connection", "close");
            request->send(response);
        } }, handleUpload);
    server.begin();
    Serial.println("HTTP Server running");
}
```

And that's it. Works like a charm first time I flushed it. Took me maybe an hour. It's not very robust but for simple firmware flush OTA is more than enough. I used it on my garage door opener code.

Levardar, 2020-10-06:
> It case it helps anybody, I implemented basic OTA based on source: https://lastminuteengineers.com/esp32-ota-web-updater-arduino-ide/
> But I am using async web server so I used the hint provided in this thread. I don't chunk the file manually or flush to SPIFFS. Instead I use Update.h library to do all hard work. Code is pretty simple:
> 
> _OTA.h_
> this is basically the form from the source mentioned, but I removed that unnecessary login, since it does not protect anything anyways.
> 
> ```
> 
> /* Style */
> String style =
> "<style>#file-input,input{width:100%;height:44px;border-radius:4px;margin:10px auto;font-size:15px}"
> "input{background:#f1f1f1;border:0;padding:0 15px}body{background:#3498db;font-family:sans-serif;font-size:14px;color:#777}"
> "#file-input{padding:0;border:1px solid #ddd;line-height:44px;text-align:left;display:block;cursor:pointer}"
> "#bar,#prgbar{background-color:#f1f1f1;border-radius:10px}#bar{background-color:#3498db;width:0%;height:10px}"
> "form{background:#fff;max-width:258px;margin:75px auto;padding:30px;border-radius:5px;text-align:center}"
> ".btn{background:#3498db;color:#fff;cursor:pointer}</style>";
> 
> /* Index page */
> String indexPage = 
> "<h1 style=\"color:white; text-align:center;\">ESP32 Garage Management</h1>" + style;
> 
> /* Firmware Page */
> String firmwarePage = 
> "<script src='https://ajax.googleapis.com/ajax/libs/jquery/3.2.1/jquery.min.js'></script>"
> "<form method='POST' action='#' enctype='multipart/form-data' id='upload_form'>"
> "<input type='file' name='update' id='file' onchange='sub(this)' style=display:none>"
> "<label id='file-input' for='file'>   Choose file...</label>"
> "<input type='submit' class=btn value='Update'>"
> "<br><br>"
> "<div id='prg'></div>"
> "<br><div id='prgbar'><div id='bar'></div></div><br></form>"
> "<script>"
> "function sub(obj){"
> "var fileName = obj.value.split('\\\\');"
> "document.getElementById('file-input').innerHTML = '   '+ fileName[fileName.length-1];"
> "};"
> "$('form').submit(function(e){"
> "e.preventDefault();"
> "var form = $('#upload_form')[0];"
> "var data = new FormData(form);"
> "$.ajax({"
> "url: '/update',"
> "type: 'POST',"
> "data: data,"
> "contentType: false,"
> "processData:false,"
> "xhr: function() {"
> "var xhr = new window.XMLHttpRequest();"
> "xhr.upload.addEventListener('progress', function(evt) {"
> "if (evt.lengthComputable) {"
> "var per = evt.loaded / evt.total;"
> "$('#prg').html('progress: ' + Math.round(per*100) + '%');"
> "$('#bar').css('width',Math.round(per*100) + '%');"
> "}"
> "}, false);"
> "return xhr;"
> "},"
> "success:function(d, s) {"
> "console.log('success!') "
> "},"
> "error: function (a, b, c) {"
> "}"
> "});"
> "});"
> "</script>" + style;
> ```
> 
> _server.cpp_
> 
> ```
> #include <ESPAsyncWebServer.h>
> #include <OTA.h>
> #include <Update.h>
> 
> AsyncWebServer server(80);
> 
> void notFoundResponse(AsyncWebServerRequest *request)
> {
>     request->send(404, "text/plain", "Not found");
> }
> 
> // handle the upload of the firmware
> void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final)
> {
>     // handle upload and update
>     if (!index)
>     {
>         Serial.printf("Update: %s\n", filename.c_str());
>         if (!Update.begin(UPDATE_SIZE_UNKNOWN))
>         { //start with max available size
>             Update.printError(Serial);
>         }
>     }
> 
>     /* flashing firmware to ESP*/
>     if (len)
>     {
>         Update.write(data, len);
>     }
> 
>     if (final)
>     {
>         if (Update.end(true))
>         { //true to set the size to the current progress
>             Serial.printf("Update Success: %ub written\nRebooting...\n", index+len);
>         }
>         else
>         {
>             Update.printError(Serial);
>         }
>     }
>     // alternative approach
>     // https://github.com/me-no-dev/ESPAsyncWebServer/issues/542#issuecomment-508489206
> }
> 
> // mains erver handling block
> void startServer()
> {
>     // web interface for OTA
>     server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
>         AsyncWebServerResponse *response = request->beginResponse(200, "text/html", indexPage);
>         response->addHeader("Connection", "close");
>         request->send(response);
>     });
>     server.on("/firmware", HTTP_GET, [](AsyncWebServerRequest *request) {
>         AsyncWebServerResponse *response = request->beginResponse(200, "text/html", firmwarePage);
>         response->addHeader("Connection", "close");
>         request->send(response);
>     });
>     // handling uploading firmware file
>     server.on("/update", HTTP_POST, [](AsyncWebServerRequest *request) {
>         if (!Update.hasError()) {
>             AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "OK");
>             response->addHeader("Connection", "close");
>             request->send(response);
>             ESP.restart();
>         } else {
>             AsyncWebServerResponse *response = request->beginResponse(500, "text/plain", "ERROR");
>             response->addHeader("Connection", "close");
>             request->send(response);
>         } }, handleUpload);
>     server.onNotFound(notFoundResponse);
>     server.begin();
>     Serial.println("HTTP Server running");
> }
> ```
> 
> And that's it. Works like a charm first time I flushed it. Took me maybe an hour. It's not very robust but for simple firmware flush OTA is more than enough. I used it on my garage door opener code.

Nice, this is what I was searching for, but just for a nice look I want to refresh the page after 100%, so the progress bar is restored to empty, the response "OK" is not received by Ajax ((console.log('success!')). Is it not working for me or is a bug in the code?

Thank's
Razvan

PPKGroup, 2021-02-09:
> It case it helps anybody, I implemented basic OTA based on source: https://lastminuteengineers.com/esp32-ota-web-updater-arduino-ide/
> But I am using async web server so I used the hint provided in [...]

Hi,
I hope all is well in your part of the world.
I'm actually contacting you in regard to a post you placed in the above.
I tried copying your OTA.h file but it seems incomplete. I get String errors when I paste it into a blank file.

In my project, I would like to be able to copy up a new firmware.bin as well as other file types like htm, css, json etc.
I believe your code may be able to handle this? Could you please share the required files and give a web browser based command line example e.g. http://mydevice.local/upload/fred.htm

me-no-dev/ESPAsyncWebServer#542

Thanks you for any help,
King regards,
Neil.
neilw@nwe.net.au

Carbicjoe, 2022-01-09:
> @atanisoft I also fixed the issue that I had with attaching the processor to that specific handler. I made it `PROGMEM` :
>
> ```
>   server.on("/files", HTTP_ANY, [](AsyncWebServerRequest *request){ [...]

Hi,
I´m looking into uploading files from a browser directly to the SD card of my ESP32. Your problem/answer seems to be 99% of what I´m looking for. Could you please share the complete solution (server side and HTML code). I can´t figure out which parts of the code snippets above are actually needed.
Thank you all for sharing this problem/solution!

06GitHub, 2023-11-25:
Thanks @OscarHanzely for this nice piece of code (OTA.h and server.cpp).

For those interested, I have merged server.cpp into "otawebupdater_espasyncwebserver.ino" file to allow straightforward compilation of ino file, see below.

Output and screenshots are also attached.

**otawebupdater_espasyncwebserver.ino**
```
// credit https://github.com/me-no-dev/ESPAsyncWebServer/issues/542#issuecomment-573445113 "Levardar commented Oct 6, 2020 It case it helps anybody, ..."
// code below :
// - reuses "OTA.h" as is
// - merges "server.cpp" into "otawebupdater_espasyncwebserver.ino" (no value added except code can be compiled as is)
// once code is compiled and uploaded on ESP32, OTA firmware update is available at url "IP/firmware", eg "http://192.168.1.27/firmware"
// Note : since IP address is displayed with ESP_LOGI function, code should be compiled with option "Core Debug Level:Info" (in tools menu) otherwise IP is not displayed
#include <WiFi.h>
#include <WiFiClient.h>

#include <ESPAsyncWebServer.h>
#include "OTA.h"               // Web page definition
#include <Update.h>

AsyncWebServer server(80);

const char* host = "esp32";
const char* ssid = "MYSSID";
const char* password = "MYPASSWORD";

void notFoundResponse(AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
}

// handle the upload of the firmware
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
    // handle upload and update
    if (!index){
        Serial.printf("Update: %s\n", filename.c_str());
        if (!Update.begin(UPDATE_SIZE_UNKNOWN)) { //start with max available size
            Update.printError(Serial);
        }
    }
    /* flashing firmware to ESP*/
    if (len) {
        Update.write(data, len);
    }

    if (final) {
        if (Update.end(true)) { //true to set the size to the current progress
            Serial.printf("Update Success: %ub written\nRebooting...\n", index+len);
        }
        else {
            Update.printError(Serial);
        }
    }
    // alternative approach
    // https://github.com/me-no-dev/ESPAsyncWebServer/issues/542#issuecomment-508489206
}

// mains server handling block
void startServer(){
    // web interface for OTA
    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        AsyncWebServerResponse *response = request->beginResponse(200, "text/html", indexPage);
        response->addHeader("Connection", "close");
        request->send(response);
    });
    server.on("/firmware", HTTP_GET, [](AsyncWebServerRequest *request) {
        AsyncWebServerResponse *response = request->beginResponse(200, "text/html", firmwarePage);
        response->addHeader("Connection", "close");
        request->send(response);
    });
    // handling uploading firmware file
    server.on("/update", HTTP_POST, [](AsyncWebServerRequest *request) {
        if (!Update.hasError()) {
            AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "OK");
            response->addHeader("Connection", "close");
            request->send(response);
            ESP.restart();
        } else {
            AsyncWebServerResponse *response = request->beginResponse(500, "text/plain", "ERROR");
            response->addHeader("Connection", "close");
            request->send(response);
        } }, handleUpload);
    server.onNotFound(notFoundResponse);
    server.begin();
    Serial.println("HTTP Server running");
}

/* setup function */
void setup() {
  delay(2000); // allow time to display IP address on Serial
  // Connect to WiFi network
  WiFi.begin(ssid, password);

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    ESP_LOGI("",".");
  }
  ESP_LOGI("","Connected to %s IP %s ", ssid, WiFi.localIP().toString().c_str());
  startServer(); // start server
}

void loop() {
  delay(1);
}
```

**output**
18:37:47.815 -> [  2752][I][otawebupdater_espasyncwebserver.ino:89] setup(): [] .
18:37:47.815 -> [  2753][I][otawebupdater_espasyncwebserver.ino:91] setup(): [] Connected to MYSSID IP 192.168.1.27 

**screenshots**
[image]
[image]
[image]