me-no-dev/ESPAsyncWebServer #543: Compatibility

MajicCat, 2019-07-05:
Looks like there is an incompatibility with core ESP (tried 1.01 and 1.02)

C:\Users\...\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'bool _start_async_task()':

C:\Users\...\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:201:141: error: 'xTaskCreateUniversal' was not declared in this scope

         xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);

atanisoft, 2019-07-05:
This method was added in arduino-esp32 1.0.2. The Arduino IDE does *NOT* always compile with the latest bits even if you tell it to.

The easiest fix is to do a clean install of the Arduino IDE.

MajicCat, 2019-07-05:
Yes that fixed it... but the existing working Ethernet calls broke.

(There was a old issue with ESP / Arduino compatibility. It may have broke again... )

C:\Program Files (x86)\Arduino\libraries\Ethernet\src/Ethernet.h:258:2: error: invalid abstract return type 'EthernetClient'
  EthernetClient available();
  ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
 class EthernetClient : public Client {
       ^

In file included from C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:157:0,
                 from sketch\B.ino.cpp:1:

C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:31:17: note: 	virtual int Client::connect(IPAddress, uint16_t, int)

     virtual int connect(IPAddress ip, uint16_t port, int timeout) =0;
                 ^

C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:32:17: note: 	virtual int Client::connect(const char*, uint16_t, int)
     virtual int connect(const char *host, uint16_t port, int timeout) =0;
                 ^

In file included from C:\Users\james\Documents\Arduino\BRIDGE\HAL.ino:6:0:

C:\Program Files (x86)\Arduino\libraries\Ethernet\src/Ethernet.h:259:2: error: invalid abstract return type 'EthernetClient'

  EthernetClient accept();

atanisoft, 2019-07-05:
That is unrelated to ESPAsyncWebServer, that would be https://github.com/espressif/arduino-esp32/issues/2786.

jjmonteiro, 2019-07-13:
I'm having same issue but "AsyncTCP.cpp:204:141" instead. Already reinstalled the IDE and still persists. Any help?

bikevit2008, 2019-07-16:
I have this trouble too

jjmonteiro, 2019-07-16:
Solved! 
We need to use latest version of the arduino-esp32 framework 1.0.3-rc1

either you install it manually from:

 https://github.com/espressif/arduino-esp32/releases/tag/1.0.3-rc1 

or you can enable pre-release versions on the Arduino IDE, using the board manager URL:

https://dl.espressif.com/dl/package\_esp32\_dev\_index.json

emry78, 2019-08-30:
> Solved!
> We need to use latest version of the arduino-esp32 framework 1.0.3-rc1
>
> either you install it manually from:
>
> https://github.com/espressif/arduino-esp32/releases/tag/1.0.3-rc1
>
> or you can [...]

HI MajicCat,

I 'm new user. can you share how install manual arduino-esp32 framework 1.0.3-rc1. I also have problem like this

jjmonteiro, 2019-08-30:
Hi emry78,

Download the zip file and replace the files on your Arduino/hardware/espressif/esp32 folder
But I'll suggest you instead to update your board manager url in the Arduino IDE and update the board version from there.

emry78, 2019-09-04:
> Hi emry78,
> 
> Download the zip file and replace the files on your Arduino/hardware/espressif/esp32 folder
> But I'll suggest you instead to update your board manager url in the Arduino IDE and update the board version from there.

Hi jjmonteiro,
Im using linux os. little bit confuse the folder. I think different from window base. i saw linux have folder like this /home/amry/.arduino15/packages/esp32/hardware/esp32/1.0.1 . So shoud be pun in this folder?. need your advise. Thanks

closed 2019-11-17 by stale[bot]