me-no-dev/ESPAsyncWebServer #544: serveStatic() triggers the watchdog and restarts ESP32.

GeorgeFlorian, 2019-07-07:
Hello !

I am trying to clean up my code and as such I am trying to replace all the GET handlers with
```
server.serveStatic("/", SPIFFS, "/").setDefaultFile("indexAP.html").setTemplateProcessor(processor);
```

Buuut, I'm having no luck. The only page that gets served is the `indexAP.html` and not even this works properly. Only a small part of the CSS works, the image/png doesn't gets served. The ESP32 also restarts after a few seconds on this page with the following errors:
```
E (8187) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8187) task_wdt:  - async_tcp (CPU 0/1)
E (8187) task_wdt: Tasks currently running:
E (8187) task_wdt: CPU 0: IDLE0
E (8187) task_wdt: CPU 1: async_tcp
E (8187) task_wdt: Aborting.
abort() was called at PC 0x400e2d0b on core 0
```

Here is the full code:
# Rewrites:
```
  server.rewrite("/events.html","/logs");
  server.rewrite("/AP_configPage.html","//IP-Config");
  server.rewrite("/dhcpIP.html","/dhcpIP");
  server.rewrite("/staticIP.html","/staticIP");
```
# serveStatic:
```
server.serveStatic("/", SPIFFS, "/").setDefaultFile("indexAP.html").setTemplateProcessor(processor);
```
# POST handler:
```
  server.on("/", HTTP_POST,[](AsyncWebServerRequest *request) {
    if(request->hasArg("register")){
      int params = request->params();
      String values_user[2];
      for(int i=0;i<params;i++){
        AsyncWebParameter* p = request->getParam(i);
        if(p->isPost()){
            logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
            values_user[i] = p->value();            
          } else {
              logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
            }
      } // for(int i=0;i<params;i++)
      
      if(values_user[0] != NULL && values_user[0].length() > 4 &&
        values_user[1] != NULL && values_user[1].length() > 7) {
            File userWrite = SPIFFS.open("/user.txt", "w");
            if(!userWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write USER credentials !");
            userWrite.println(values_user[0]);  // Username
            userWrite.println(values_user[1]);  // Password
            userWrite.close();
            logOutput("Username and password saved !");          
            request->redirect("/IP-Config");
            // digitalWrite(LED, LOW);
      } else request->redirect("/register");  
    } else if (request->hasArg("skip")) {
      request->redirect("/IP-Config");
    } else if(request->hasArg("import")) {
      request->redirect("/files");
    } else if(request->hasArg("dhcp")) {
      request->redirect("/dhcpIP");
    } else if(request->hasArg("static")) {
      request->redirect("/staticIP");
    } else if(request->hasArg("saveDHCP")) {
        int params = request->params();
        String values_dhcp[2];
        for(int i=0;i<params;i++){
          AsyncWebParameter* p = request->getParam(i);
[... 244 lines cut ...]
    int params = request->params();
    String values_dhcp[2];
    for(int i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isPost()){
          logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
          values_dhcp[i] = p->value();            
        } else {
            logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
          }
    }
    if(values_dhcp[0] != NULL && values_dhcp[0].length() != 0 &&
      values_dhcp[1] != NULL && values_dhcp[1].length() != 0) {
      File inputsWrite = SPIFFS.open("/network.txt", "w");
      if(!inputsWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write DHCP IP credentials !");
      inputsWrite.println(values_dhcp[0]);  // SSID
      inputsWrite.println(values_dhcp[1]);  // Password
      inputsWrite.close();
      logOutput("Configuration saved !");
      request->redirect("/logs");
      shouldReboot = true;
    } else request->redirect("/dhcpIP");
  
  }); // server.on("/dhcpLogin", HTTP_POST, [](AsyncWebServerRequest * request)
```

atanisoft, 2019-07-07:
@GeorgeFlorian I think you have a typo...
```
server.rewrite("/AP_configPage.html","//IP-Config");
```
Shouldn't there only be one / in the target?

Also please capture the console output at the VERBOSE log settings so @me-no-dev can help investigate why async_tcp is triggering the WDT.

GeorgeFlorian, 2019-07-07:
> @GeorgeFlorian I think you have a typo...
> 
> ```
> server.rewrite("/AP_configPage.html","//IP-Config");
> ```
> 
> Shouldn't there only be one / in the target?
> 
> Also please capture the console output at the VERBOSE log settings so @me-no-dev can help investigate why async_tcp is triggering the WDT.

Which one is correct:
`server.rewrite("/IP-Config","/AP_configPage.html");` or `server.rewrite("/AP_configPage.html","/IP-Config");` ?
I am thinking that I want to send from `.html` to the uri so I'll chose: 

The typo is only in this thread. After I've pasted the code here I removed one of the back slash.

That was the VERBOSE log:
```
E (21547) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21547) task_wdt:  - async_tcp (CPU 0/1)
E (21547) task_wdt: Tasks currently running:
E (21547) task_wdt: CPU 0: IDLE0
E (21547) task_wdt: CPU 1: async_tcp
E (21547) task_wdt: Aborting.
abort() was called at PC 0x400e2ec3 on core 0

Backtrace: 0x4008c7e0:0x3ffbe160 0x4008ca11:0x3ffbe180 0x400e2ec3:0x3ffbe1a0 0x40081789:0x3ffbe1c0 0x4014f627:0x3ffbc320 0x400e1523:0x3ffbc340 0x4008a67d:0x3ffbc360 0x4008877d:0x3ffbc380
```

me-no-dev, 2019-07-07:
you have cases where you send two responses/redirects or not at all. Any time you get WDT, means it's your code that is bad :)

velbn, 2019-07-08:
Hi 
I have a too similar issue. The issue started when I updated ESPAsyncWebServer
[image]
is there anyway to backtrace this to corresponding request?

velbn, 2019-07-08:
I think it has something to do with the partition size of the SPIFFS.
When I make the partition size smaller. It goes first of all much faster but also more stable

vinot, 2019-07-15:
Hi,

try add:
#include "esp_wifi.h"
 esp_wifi_set_ps (WIFI_PS_NONE);

and test, it solved my problem

GeorgeFlorian, 2019-07-16:
> Hi,
> 
> try add:
> #include "esp_wifi.h"
> esp_wifi_set_ps (WIFI_PS_NONE);
> 
> and test, it solved my problem

Will try.

closed 2019-10-05 by stale[bot]

d0ct0rd-dalimo, 2019-10-28:
It didn't help me, I have the same problem, but with AsyncWebServer

GeorgeFlorian, 2019-10-28:
> It didn't help me, I have the same problem, but with AsyncWebServer

What problems ? I advise you to open a separate issue.