me-no-dev/ESPAsyncWebServer #553: How can I use `AsyncClient *client = new AsyncClient();` with a simple `AsyncWebServer server(80);` ?

GeorgeFlorian, 2019-07-15:
Hello !

I would like to gain access to `client.onTimeout([](void*, AsyncClient*, uint32_t time) {} );` but I don't know how to use `AsyncClient *client = new AsyncClient();`.

I don't have a WebSocket in my project, nor do I understand it's purpose/use.

I've tried adding `AsyncClient *client = new AsyncClient();` outside of `void setup() {}` but when using `Serial.println(aClient->connected());` or `Serial.println(aClient->localIP().toString());` I always get 0 or 0.0.0.0. It's like I don't have any client.

This is surely because I don't attach the `client` to nothing.

How can I use this ? There isn't any example.

philbowles, 2019-08-16:
??? the AsyncClient *client = new AsyncClient(); needs to be in a _different sketch_ on a _different machine_ . It needs to be in the code that connects to this server, not in this server itself.

AsyncClient is used by code that requests a TCP connection to a server, i.e. it is a client (hence the name). I looked at your related issue. It appears to me you have a fundamental misunderstanding of the client-server model and/or what code runs where...

We are talking about two totally separate MCUs here

CLIENT -------------------------------------------->SERVER
creates AsyncClient.......................................................runs AsyncWebserver
connects to: ------------------------------------>
<------------------------------------------------- replies to client
asks for data (usually web page)--------------->
<------------------------------------------------ sends web page or 404 etc

Both of these objects  happen to use the underlying asyncTCP library (each with its own copy in its own mcu), one to send, one to receive, don't mix up the different functionality just because they share a common library

Otherwise, what am I missing? One of us is very confused.

AkhileshThorat, 2019-10-01:
Hi @philbowles 
I think the problem @GeorgeFlorian encounters is the way to connect to a remote webSocket server(different ESP32). As he said there aren't any examples demonstrating how to connect a Client(first ESP32 ) to a server(second esp32 running async web server) and vice versa.
Even I'm unable to figure out the API's for connecting to a remote server.
Lets say my server is running on http://{ESP-IP}/ws, what should I write to establish a connection with this endpoint?

closed 2019-12-14 by stale[bot]

jhoughjr, 2020-01-29:
I also have found the proper way to make a client connection, and I really need that functionality. I'll keep working on it and if I figure it out, I'll post it here.

GeorgeFlorian, 2020-01-29:
> I also have found the proper way to make a client connection, and I really need that functionality. I'll keep working on it and if I figure it out, I'll post it here.

Great