me-no-dev/ESPAsyncWebServer #555: ESP32 freezes out of nowhere. Destination Host Unreachable

GeorgeFlorian, 2019-07-18:
OS: Linux Mint 19.1
Board: ESP32-EVB
IDE: PlatformIO IDE
`platform = https://github.com/platformio/platform-espressif32.git`

Hello !

I am using ESPAsyncWebServer to serve some web-pages and to control the 2 relays that are on the board.

The issue I am having is that the ESP freezes randomly with no DEBUG message. I have set the DEBUG level at Verbose: `build_flags = -DCORE_DEBUG_LEVEL=5`.

I am also using the following methods in order to prevent the ESP from freezing, but they don't help at all. With or without them I can't make the project reliable.
```
  enableCore1WDT();
  esp_wifi_set_ps (WIFI_PS_NONE);
```

I get the following browser error:
```
This site can’t be reached
ERR_ADDRESS_UNREACHABLE
```

And pinging the ESP ends with `Destination Host Unreachable`.

The issue I am experiencing makes the ESP a very unreliable board. I don't know if it's the ESP at fault or the Server's Library.

The `void loop()` is pretty straight forward.
```
void loop(){
  if (shouldReboot) {
    logOutput("Restarting in 5 seconds...");
    delay(5000);
    server.reset();
    ESP.restart();
  }
  currentTimeRelayOne = millis();
  currentTimeRelayTwo = millis();

  if(startTimeRelayOne != 0 && !needManualCloseRelayOne) {
    if ((currentTimeRelayOne - startTimeRelayOne) > (Delay1.toInt()*1000)) {
      digitalWrite(RELAY1,LOW);
      status1 = "OFF";
      logOutput((String)relay1 + " closed");
      startTimeRelayOne = 0;
    }
  }

  if(startTimeRelayTwo != 0 && !needManualCloseRelayTwo) {
    if ((currentTimeRelayTwo - startTimeRelayTwo) > (Delay2.toInt()*1000)) {
      digitalWrite(RELAY2,LOW);
      status2 = "OFF";
      logOutput((String)relay2 + " closed");
      startTimeRelayTwo = 0;
    }
  }
  delay(2);
}
```

vinot, 2019-07-19:
HI,

I use the library on esp8266 and esp32 and it works perfect. Try it step by step (first, only connecting to network and ping) and don't use delay() as told on startpage (with esp8266 I don't saw problems with delay() but not with esp32)

cheers

toni

GeorgeFlorian, 2019-07-19:
> HI,
> 
> I use the library on esp8266 and esp32 and it works perfect. Try it step by step (first, only connecting to network and ping) and don't use delay() as told on startpage (with esp8266 I don't saw problems with delay() but not with esp32)
> 
> cheers
> 
> toni

I don't understand your comment.
Try what step by step ?
I did try it step by step... 4 months ago when I first picked it up.

vinot, 2019-07-19:
Hi,

try first with any close example and test anything before go beyond. And second, as state in

https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember

 eliminate any delay and keep loop(){}; empty ;)

toni

GeorgeFlorian, 2019-07-20:
> Hi,
> 
> try first with any close example and test anything before go beyond. And second, as state in
> 
> https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember
> 
> eliminate any delay and keep loop(){}; empty ;)
> 
> toni

Hello !

I don't have any delays in any handler.

Secondly, I don't understand the point of these tests. As I said, I started small 4 months ago. And it doesn't matter if it works with simple projects.

I need it to work with my current project which is pretty ample.

vinot, 2019-07-20:
Hi,

if I understand well, this is part of your code:

void loop(){
  if (shouldReboot) {
    logOutput("Restarting in 5 seconds...");
    delay(5000);
    server.reset();
    ESP.restart();
  }
  currentTimeRelayOne = millis();
  currentTimeRelayTwo = millis();

  if(startTimeRelayOne != 0 && !needManualCloseRelayOne) {
    if ((currentTimeRelayOne - startTimeRelayOne) > (Delay1.toInt()*1000)) {
      digitalWrite(RELAY1,LOW);
      status1 = "OFF";
      logOutput((String)relay1 + " closed");
      startTimeRelayOne = 0;
    }
  }

  if(startTimeRelayTwo != 0 && !needManualCloseRelayTwo) {
    if ((currentTimeRelayTwo - startTimeRelayTwo) > (Delay2.toInt()*1000)) {
      digitalWrite(RELAY2,LOW);
      status2 = "OFF";
      logOutput((String)relay2 + " closed");
      startTimeRelayTwo = 0;
    }
  }
  delay(2);
}

isn't it? There are two delays() and the whole code is inside loop() !!

GeorgeFlorian, 2019-07-20:
Yes. You saw that correctly. But @me-no-dev said to not put delays INSIDE a handler. The handler will be inside `void setup()`.

This is a handler:
```
  server.on("/index", HTTP_ANY, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.htm");
  });
```

You can see in the documentation that you can use whatever you want in `void loop()`.

In this example: https://github.com/me-no-dev/ESPAsyncWebServer#setting-up-the-server
we have:
```
void loop(){
  if(shouldReboot){
    Serial.println("Rebooting...");
    delay(100);
    ESP.restart();
  }
  static char temp[128];
  sprintf(temp, "Seconds since boot: %u", millis()/1000);
  events.send(temp, "time"); //send event "time"
}
```
Judging by this we can safely assume that it's okay to use `delay()` whenever we want to, outside the Server's handlers.

closed 2019-10-05 by stale[bot]