me-no-dev/ESPAsyncWebServer #556: [Question] serveStatic() usage with gzipped content

AlexFundorin, 2019-07-19:
I'm currently using the following method to serve files from SPIFFS partition. 

```
    server.on("/", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse(SPIFFS, "/index.html.gz", "text/html");
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
```

It's VERY slow, even with <10Kb files.

Somebody told me that I should use serveStatic() method, instead.
My question is how do I add "gzip" header to this string, so that the server can tell the browser, that the content is gzipped?

```
server.serveStatic("/index.html", SPIFFS, "/index.html.gz");
```

Pablo2048, 2019-07-19:
You don't need to do so - the webserver automagically serve .gz files first - see https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebHandlers.cpp#L116 , https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebHandlers.cpp#L142

AlexFundorin, 2019-07-19:
Yes. I figured that out, yesterday. For some reason, web page loading time is still slow.
Should I add cache control or some other handlers?

maxdd, 2019-10-01:
I'm also interested in this, i'll give it a try and report a feedback if i also have bad performance.
Can you perhaps check with the browser developer console where the bad performance lies? Is it before/during/after the gzip is served?

closed 2019-12-14 by stale[bot]

patrickzzz, 2024-10-28:
> You don't need to do so - the webserver automagically serve .gz files first - see https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebHandlers.cpp#L116 , https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebHandlers.cpp#L142

Although, as "_gzipFirst" is uncertain, if you want to force the .gz file, you have to remove the .html or .js file..
Also the Accept-Encoding Header doesnt seem to be respected!

But nevertheless in situations where this is no problem, this way is nice!

Regards,
Patrick