me-no-dev/ESPAsyncWebServer #558: Can not load index.html from SPIFFS using ESP32

williamesp2015, 2019-07-20:
I am using ESP32 on PlatoformIO. Websocvket is working but can not load index.html by typing APIP on chrome or other browsers.
void setup()
{
.......
WiFi.mode(WIFI_MODE_APSTA);
delay(100);
Serial.print("Setting soft-AP configuration ... ");
Serial.println(WiFi.softAPConfig(local_IP, gateway, subnet) ? "Ready" : "Failed!");
delay(100);
Serial.print("Setting soft-AP ... ");
WiFi.softAP("Captive Portal");
delay(500);
Serial.print("Soft-AP IP address = ");
Serial.println(WiFi.softAPIP());
ws.onEvent(onWsEvent);
server.addHandler(&ws);
server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html").setCacheControl("max-age=600");
server.serveStatic("/", SPIFFS, "/").setDefaultFile("app.js").setCacheControl("max-age=600");
server.begin();
}
void loop(){
  delay(100);
}

atanisoft, 2019-07-20:
Your second call to serveStatic is overriding the first one for the default file. If you remove the second call to serveStatic it will likely work as you expect.

williamesp2015, 2019-07-21:
@atanisoft . Thank you. Unfortunately still does not working.

brunohorta82, 2019-08-07:
same problem

Blend3rman, 2019-10-18:
I have solved this problem. Working on an ESP8266-EX, I tried to run the sketch on the main page as-is, and ran into the same issues. 
@williamesp2015 @brunohorta82 try including the following before you try to serve from SPIFFS:
```
  if(!SPIFFS.begin()) {
    Serial.println("Mount SPIFFS failed, exiting...");
    return;
  }
  Serial.println("Mount SPIFFS successfully.");
```

IMO the example sketch needs to be modified to include a call to `SPIFFS.begin()`

07fi123, 2019-10-20:
>   if(!SPIFFS.begin()) {
>     Serial.println("Mount SPIFFS failed, exiting...");
>     return;
>   }
```
SPIFFS.begin();
//other code
server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");
server.begin();

```
This worked for me! Thanks @Blend3rman !

closed 2020-01-02 by stale[bot]