me-no-dev/ESPAsyncWebServer #565: ESP32-Captive portal doesn't start automatically by calling index.html from SPIFFS

Theghost90, 2019-07-26:
Captive portal doesn't start automatically as it should be. Indeed, below are the issues:

**### 1. The esp doesn't broadcast automatically his his invitation to access to the web portal
2. It doesn't load the root file (index.html) from the SPIFFS.**

All my filles are located on the SPIFFS and loaded from SPIFFS.
index.html is the login page, but that should be automatically loaded when a client connects to AP.

aliyehia92, 2019-08-27:
Perhaps share some code snippets?

Theghost90, 2019-08-27:
Below is the code:

#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <DNSServer.h>
#include <SPIFFS.h>
#include <HardwareSerial.h>
#include <PZEM004T.h>

const byte DNS_PORT = 53;
const char *espssid = "HUAWEI P9";
const char *esppassword = "On@mbele_1989";

AsyncWebServer server(80);
DNSServer dnsServer;
IPAddress local(192, 168, 4, 25);
IPAddress gateway(192, 168, 4, 0);
IPAddress netMsk(255, 255, 255, 0);
int max_connection = 4;

HardwareSerial PzemSerial2(2);     // Use hwserial UART2 at pins IO-16 (RX2) and IO-17 (TX2)
PZEM004T pzem(&PzemSerial2);
IPAddress ip(192,168,1,1); 

/*String getPZEMCurrent(){
  float i = pzem.current(ip);
   if(i >= 0.0){ Serial.print(i);Serial.print("A; "); return String(i); }
  }
String getPZEMVoltage(){
  float v = pzem.voltage(ip);
  if (v >= 0.0){Serial.print(v);Serial.print("V; "); return String(v);}
  }
String getPZEMPower(){
  float p = pzem.power(ip);
   if(p >= 0.0){ Serial.print(p);Serial.print("W; "); return String(p);}
  }
String getPZEMEnergy(){
  float e = pzem.energy(ip);
   if(e >= 0.0){ Serial.print(e);Serial.print("Wh; "); return String(e);}
  }*/
void handleLogin(AsyncWebServerRequest *request){
  if(request->hasParam("USERNAME") && request->hasParam("PASSWORD")){

String username = request->getParam("USERNAME")->value().c_str();
String password = request->getParam("PASSWORD")->value().c_str();

     if ( username == "john" && password == "pass123") {
        File file = SPIFFS.open("/dashboard.html", "r");
        request->send(SPIFFS, "/dashboard.html", "text/html");
        file.close();
      Serial.println("Log in Successful");
     }
     else {   
        File file = SPIFFS.open("/indexerror.html", "r");
        request->send(SPIFFS, "/indexerror.html", "text/html");
        file.close();
      Serial.println("Log in Failed, Wrong Username/Password");
      }
}
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pzem.setAddress(ip);
  
  if(!SPIFFS.begin()){
    Serial.println("SPIFFS Mount failed");
  } else {
    Serial.println("SPIFFS Mount succesfull");
  }
  
    WiFi.mode(WIFI_AP);
  Serial.println();
Serial.println("Configuring access point...");
WiFi.softAPConfig(local, gateway, netMsk);
  WiFi.softAP(espssid, esppassword, max_connection);
  delay(500);
  Serial.print("Device ");
  Serial.print(espssid);
  Serial.println(" started");
  Serial.print("Adresse IP: ");
  Serial.println(WiFi.softAPIP());

  if(!dnsServer.start(DNS_PORT, "*", WiFi.softAPIP()))
    Serial.printf("\n failed to start dns service \n");
  
  //redirect all traffic to index.html
  server.onNotFound([](AsyncWebServerRequest *request) {
      const char *metaRefreshStr = "<head><meta http-equiv=\"refresh\" content=\"0; url=http://config.local/dashboard.html.html\" /></head><body><p>redirecting...</p></body>";
      request->send(SPIFFS, "/index.html", metaRefreshStr);
 //i can use request->redirect(SPIFFS, "/index.html");
 //external url: request->redirect("http://ewsp.com");
  });
/*server.on("/generate_204", HTTP_GET, [](AsyncWebServerRequest *request){
    request->redirect("/index.html");
    delay(100);
  });*/
  server.on("/dashboard.html", handleLogin);
  server.on("/indexerror.html", handleLogin);
/*  server.on("/CurrentValues", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", getPZEMCurrent().c_str());
  });
  server.on("/VoltageValues", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", getPZEMVoltage().c_str());
  });
  server.on("/PowerValues", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", getPZEMPower().c_str());
  });
  server.on("/EnergyValues", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", getPZEMEnergy().c_str());
  });*/
  server.serveStatic("/css", SPIFFS, "/css");
  server.serveStatic("/img", SPIFFS, "/img");
  server.serveStatic("/", SPIFFS, "/index.html");
  server.serveStatic("/", SPIFFS, "/dashboard.html");
  server.serveStatic("/", SPIFFS, "/indexerror.html");
  server.serveStatic("/js", SPIFFS, "/js");

  server.begin();
  Serial.println("Server started successfully");
}

void loop() {
  // put your main code here, to run repeatedly:
  dnsServer.processNextRequest();
}

Saur0o0n, 2019-09-05:
How's that suppose to work?
```
server.serveStatic("/", SPIFFS, "/index.html");
server.serveStatic("/", SPIFFS, "/dashboard.html");
server.serveStatic("/", SPIFFS, "/indexerror.html");
```

Theghost90, 2019-09-07:
Please what's supposed to be done?
Thanks to correct me.
Because I just copied a little from Synchronous webserver code using SPIFFS

closed 2019-11-20 by stale[bot]