me-no-dev/ESPAsyncWebServer #569: response stream gets slower and slower with every request

aliyehia92, 2019-08-07:
Hello everyone.

I'm gonna get right into it. Below is my very simple code to try out **AsyncResponseStream**. 

```
server.on(file_name, HTTP_POST, [file_name](AsyncWebServerRequest * request) {
     if(strcmp(file_name,"/exportreports") == 0) {
           AsyncResponseStream *response = request->beginResponseStream("text/html");
     }
}
```

That's it. I'm waiting for a POST request from the client, if the requested resource is "/exportreports" I'm just initializing an **AsyncReponseStream** object and not really doing anything with it. However, with each POST request received the reply time gets ever so slightly slower until ESP32 resets and connects again. I've also noticed that every time it takes the same exact amount of requests for the ESP to reset, which is 95.

To make sure that it wasn't my code that was faulty, I removed 

`AsyncResponseStream *response = request->beginResponseStream("text/html");`

and replaced it with 

`request->send(200, "text/plain", "success");`

and just like that the issue was gone and the response time remained the same and the ESP32 didn't reset no matter how many requests it received.

I tried playing with the buffer size, something like this:
`AsyncResponseStream *response = request->beginResponseStream("text/html",100);`

This only resulted in increasing the number of attempts required to eventually cause the ESP to reset. I also tried setting response to NULL after the above line of code and it did not make a difference. I'm pretty sure there's a buffer filling up somewhere but I couldn't put my hand on it.

atanisoft, 2019-08-07:
@aliyehia92 your creation of that response object without sending it is causing the esp32 to run out of memory likely. Check the serial console and you likely will see an exception when it reboots about heap.

aliyehia92, 2019-08-12:
@atanisoft thanks! I can't believe I didn't take this into consideration.. I'd been trying to solve this for 2 days. Solved!

closed 2019-08-12 by aliyehia92