me-no-dev/ESPAsyncWebServer #574: File download to Android Phone from SPIFFS.

zekageri, 2019-08-15:
The file download is working fine on Windows like this:

```
server.on("/errorLog", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, ErrorLog, "text/html", true);
  });
```
This is downloading the "ErrorLog.txt" to windows, but not on android. 
The android says just only that the download is failed. No error code and nothing.

philbowles, 2019-08-16:
Hmmm. I presume ErrorLog is a String holding the value "ErrorLog.txt" but then I get confused. Is the file text or html? Can you show us an example of whats in it? The reply says its ""text/html" - if it is NOT valid html then different browsers (note NOT OS..."to windows" and "the android" are both incorrect and irrelevant) may well handle the mismatch differently.

The author would need to know _which browser_ on windows, and _which browser_ on android and also what's actually in the file. Without that information there is nothing that could be done, given the minimal amount of information and zero error messages / output you have provided.

zekageri, 2019-08-16:
Thanks you for your answer. There are no error messages. The file is a simple text file in the esp32's SPIFFS in .txt format. In Windows the file gets downloaded without a doubt, using any browser. In Android, when i request the file there is just one "error" that said that the file cannot be downloaded. Iam using async tcp server.

philbowles, 2019-08-17:
"The file is a simple text file" then your code is bad. The MIME type should always match the true nature of the file content for _any_ browser to successfully render / handle the file.

You should reply with type "text/plain" and NOT "text/html". Also: "...using any browser" really? which ones did you try?

"Android" is not a browser. Which browser were you using?

In my first attempt to help you I said "...also what's actually in the file." Your reply "The file contains just plain text, nothing else" is NOT a helpful answer. Unless you can show _the contents of the file_ AND name the browser(s) and version of each, then I can't help you any more.

zekageri, 2019-08-17:
You are very edgy dont you?
I tought this is obvious that the plain text means plain text like: hahsndjb
When i said with any browser I mean any browser. When i tried to download
it on windows with Chrome, firefox, ie, opera they are all working and I
get my general .txt file downloaded without a problem. The file contains
String or whatever characters that are usually in a simple text file. On
Android ofc I tried this browsers too, including the basic Android browser.

Phil Bowles <notifications@github.com> ezt írta (időpont: 2019. aug. 17.,
Szo 13:53):

philbowles, 2019-08-17:
I'm not "edgy" , I just don't have the patience for people who repeatedly refuse to provide the necessary information required to solve their problem - and then argue with people trying to help them - for free and out of the goodness of their heart.

Twice now I have told you what is required. I'll do it one more time: "You should reply with type "text/plain" and NOT "text/html".

What happened when you tried it? You DID try it didn't you?

Also if after twice being told that you need to provide the CONTENTS OF THE TEXT FILE you still haven't done so...is it because you don't understand what is being asked? Is it because you don't know how to? I wanted to see _whats in the file_ 

If you don't want my help, that's fine. I'm done with this pointless conversation.

closed 2019-08-26 by zekageri