me-no-dev/ESPAsyncWebServer #576: async_tcp / SPIFFSEditor triggers watchdog

charlie71, 2019-08-16:
I have uploaded the ESP_AsyncFSBrowser example.
But when I use the spiffs editor for uploading files into the SPIFFs, the ESP32 chrashes with the following error msg:
E (77268) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

`E (77268) task_wdt:  - async_tcp (CPU 0/1)
E (77268) task_wdt: Tasks currently running:
E (77268) task_wdt: CPU 0: ipc0
E (77268) task_wdt: CPU 1: loopTask
E (77268) task_wdt: Aborting.
abort() was called at PC 0x400f4247 on core 0`

thx for help

charlie71, 2019-09-02:
I have solved the issue. 
I’m using an ESP32 with 16MB Flash und SPIFFS. Unfortunately the SPIFFS is very in performant using FAT solve this issue.

closed 2019-09-02 by charlie71

Meekdai, 2020-04-21:
Hi @charlie71 
I encountered the same problem, can you tell me the solution

charlie71, 2020-04-21:
Hi @Meekdai 
using FAT file system instead of SPIFFS solved my issue.

Meekdai, 2020-04-22:
Thanks @charlie71 , I am also using Sqlite3 for data storage, it does not seem to support the FAT file system. What caused SPIFFS to fail to use SPIFFSEditor?

charlie71, 2020-04-22:
@Meekdai 

SQLLite: I do not think that Splite3 requires SPIFFS, there is no specific reference in the library.
The Issue: It seems that some actions on the SPIFFS file system taking too long time and therefore the watchdog drops in.

Meekdai, 2020-04-23:
Great, SPIFFSEditor works well after I replaced SPIFFS to FFAT. 
But there is a problem with my Splite3, maybe I need to use sqlite3_open_v2 to configure my VFS, I will try it further. Thank you very much for your help.