me-no-dev/ESPAsyncWebServer #578: Is there a way to brake/abort file upload from server side?

Saur0o0n, 2019-08-22:
Hi,

I'm trying to prevent uploading files larger then allowed limit. But If (the only idea I had) respond from handleUpload() with request->send(200, "text/html","error"); - upload is still in progress.

Is there a way to brake accepted upload request?
```C
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
static File newFile;
String tmp=PRG_DIRECTORY;

  tmp.concat(filename.c_str());

  // Checking how much has been uploaded - if more then max_prog_size - abort
  if(len+index>max_prog_size){
     DBG Serial.println("Uploaded file too large! Aborting");
     if(newFile){
        newFile.flush();
        newFile.close();
        DBG Serial.printf("Deleting uploaded file: %s\n",tmp.c_str());
        SPIFFS.remove(tmp.c_str());
        generate_index(); // Just in case user wanted to overwrite exsisting file
     }
     request->send(200, "text/html", "<html><body><h1>File is too large!</h1> Current limit is "+String(max_prog_size)+"</body></html");
     return;
  }
    
  if(!index){
     usual stuff
  }
  for(size_t i=0; i<len; i++){
     usual stuff
  }
  if(final){
     usual stuff
     request->redirect("/programs");
  }
}

void setup(){
...
  // Upload new programs
  server.on("/upload", HTTP_POST, [](AsyncWebServerRequest *request){
    request->send(200);
  }, handleUpload);
...
}
```

atanisoft, 2019-08-22:
perhaps request->client()->abort() ? This should generate an error response on the client side automatically. I don't know of any other way to abort the upload.

Saur0o0n, 2019-08-22:
Yep, this works and aborts upload - but in a quite bad manner "(104) Connection reset by peer". Any idea how to inform client what's going on before or do it in other way?
It's not something terribly bad - since I do the check in JS on page and then in metadata file size - this is just a last resort.

atanisoft, 2019-08-22:
you might be able to call request->send(xxxx) and then request->client()->abort() after send() returns. But outside of that I can't think of any way to abort it.

Saur0o0n, 2019-08-22:
nope - I've checked that
```
     request->send(200, "text/html", "<html><body><h1>File is too large!</h1> Current limit is "+String(max_prog_size)+"</body></html");
     request->client()->abort();
```
This still gives only "Connection reset by peer".

Saur0o0n, 2019-08-22:
Ok got it - it will work, if you abort in next iteration of the function.
```C
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
static File newFile;
static boolean abort=false;
String tmp=PRG_DIRECTORY;

  if(abort){
    abort=false;
    request->client()->abort();
    return;
  }
  tmp.concat(filename.c_str());

  // Checking how much has been uploaded - if more then max_prog_size - abort
  if(len+index>max_prog_size){
     DBG Serial.println("Uploaded file too large! Aborting");
     if(newFile){
        newFile.flush();
        newFile.close();
        DBG Serial.printf("Deleting uploaded file: %s\n",tmp.c_str());
        SPIFFS.remove(tmp.c_str());
        generate_index(); // Just in case user wanted to overwrite exsisting file
     }
     request->send(200, "text/html", "<html><body><h1>File is too large!</h1> Current limit is "+String(max_prog_size)+"</body></html");
     abort=true;
     return;
  }
    
  if(!index){
...
```

closed 2019-08-22 by Saur0o0n