me-no-dev/ESPAsyncWebServer #581: HTTP_POST request showing 0 param length

chauhannaman98, 2019-08-24:
I am sending some JSON from my Postman to the ESP8266(NodeMCU) board. The param value in the serial monitor is shown as 0. I have tested it with JSON as well as plain text but the output is always 0.

Code:
```#include "ESPAsyncTCP.h"
#include "ESPAsyncWebServer.h"

const char* ssid = "ssid";
const char* password =  "pass";

AsyncWebServer server(80);

void setup() {

  Serial.begin(115200);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println(WiFi.localIP());

  server.on("/trigger_command", HTTP_POST, [](AsyncWebServerRequest *request){
 
    int paramsNr = request->params();
    Serial.println(paramsNr);
 
    for(int i=0;i<paramsNr;i++){
 
        AsyncWebParameter* p = request->getParam(i);
        Serial.print("Param name: ");
        Serial.println(p->name());
        Serial.print("Param value: ");
        Serial.println(p->value());
        Serial.println("------");
    }
 
    request->send(200, "text/plain", "message received");
  });

  server.begin();
}

void loop() {
    
}
```

**Postman Screenshot**
[image]

**Serial Monitor**
[image]

Saur0o0n, 2019-09-05:
The code looks ok to me - try to use browser to send the POST, perhaps you are doing something wrong on postman side (never used it).

closed 2019-11-18 by stale[bot]

ph1p, 2022-11-03:
Send the data like this:

```javascript
let formData = new FormData();
formData.append('user', 'name');

fetch('http://192....', {
  method: 'POST',
  body: formData,
});
```