me-no-dev/ESPAsyncWebServer #589: OTA crashing on AsyncWebServer

moothyknight, 2019-09-04:
Serial output after uploading .ino.bin file:
Update
Progress: 0%
abort() was called at PC 0x4016532d on core 1

Backtrace: 0x400926ec:0x3ffe7fd0 0x4009291d:0x3ffe7ff0 0x4016532d:0x3ffe8010 0x40082bcf:0x3ffe8030 0x400e4009:0x3ffe8050 0x400dd8ee:0x3ffe8070 0x400ddaad:0x3ffe8090 0x400d46ba:0x3ffe80b0 0x400d4829:0x3ffe8100 0x401d99a3:0x3ffe8120 0x401d982a:0x3ffe8170 0x400d846a:0x3ffe81a0 0x400d9d42:0x3ffe8200 0x400d9f1d:0x3ffe8240 0x400dcaa1:0x3ffe8260 0x400dcb21:0x3ffe82a0 0x400dd12a:0x3ffe82c0 0x4008e615:0x3ffe82f0

Rebooting...
ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4

Decoding 18 results
0x400926ec: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 707
0x4009291d: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 707
0x401646f9: is_safe_write_address at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 126
0x40082bcf: spi_flash_erase_sector at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 204 (discriminator 2)
0x400e3339: EspClass::flashEraseSector(unsigned int) at C:\Users\Joshua\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Esp.cpp line 210
0x400dcc1e: UpdateClass::_writeBuffer() at C:\Users\Joshua\Documents\Arduino\hardware\espressif\esp32\libraries\Update\src/Updater.cpp line 360
0x400dcddd: UpdateClass::write(unsigned char*, unsigned int) at C:\Users\Joshua\Documents\Arduino\hardware\espressif\esp32\libraries\Update\src/Updater.cpp line 360
0x400d45c2: handleDoUpdate(AsyncWebServerRequest*, String const&, unsigned int, unsigned char*, unsigned int, bool) at C:\Users\Joshua\AppData\Local\Temp\arduino_build_564701\sketch/WiFi_API.h line 298
0x400d4735: std::_Function_handler ::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&, String const&, unsigned int&&, unsigned char*&&, AsyncWebServerRequest*&&, bool&&) at C:\Users\Joshua\AppData\Local\Temp\arduino_build_564701\sketch/WiFi_API.h line 425
:  (inlined by) _M_invoke at c:\users\joshua\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 1871
0x401d8d17: std::function ::operator()(AsyncWebServerRequest*, String const&, unsigned int, unsigned char*, unsigned int, bool) const at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h line 73
:  (inlined by) AsyncCallbackWebHandler::handleUpload(AsyncWebServerRequest*, String const&, unsigned int, unsigned char*, unsigned int, bool) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebHandlerImpl.h line 107
0x401d8bd6: AsyncWebServerRequest::_handleUploadByte(unsigned char, bool) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebRequest.cpp line 990
0x400d8376: AsyncWebServerRequest::_parseMultipartPostByte(unsigned char, bool) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebRequest.cpp line 990
0x400d9b2e: AsyncWebServerRequest::_onData(void*, unsigned int) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebRequest.cpp line 990
0x400d9d09: std::_Function_handler ::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at C:\Users\Joshua\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebRequest.cpp line 990
:  (inlined by) _M_invoke at c:\users\joshua\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 1871
0x400dbdd1: std::function ::operator()(void*, AsyncClient*, void*, unsigned int) const at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 983
:  (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 878
0x400dbe51: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 983
0x400dc45a: _async_service_task(void*) at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 983
:  (inlined by) _async_service_task at C:\Users\Joshua\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp line 180
0x4008e615: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 355 (discriminator 1)

This is using a custom sized partition from the default and includes the BLE libraries, though they are not active while the WiFi is active. OTA worked before but after I changed partition size and added the BLE support it seems to be crashing, any ideas?

atanisoft, 2019-09-04:
This has less to do with AsyncWebServer than it does with an error being raised by the underlying ota flash write code, most likely due to attempting to write beyond the end of flash partition or possibly an invalid partition.

This should be moved to the arduino-esp32 repo and have some code attached for review so someone can help you isolate the problem.

moothyknight, 2019-09-05:
@atanisoft 

Kk, I moved it here with a code sample and a description, my full code is about 1600 lines and takes up 1.4MB so I just put the bare essentials in: https://github.com/espressif/arduino-esp32/issues/3182

closed 2019-09-05 by moothyknight