me-no-dev/ESPAsyncWebServer #590: servestatic from "/sdcard/" looks for "/sdcard/sdcard/"

brettbeeson, 2019-09-09:
Hi! I'd like to serve static files at the root uri from the file system path of "/sdcard/". So, for example, I want http://cool.beans/index.html to serve /sdcard/index.html. Readings the docs I tried:

`this->serveStatic("", SD_MMC, "/sdcard/"); // .setDefaultFile("index.htm");`

but got 404 as shown with ESP_LOG showing:

> [E][vfs_api.cpp:64] open(): /sdcard/sdcard/index.htm does not exist

[This issue is simliar](https://github.com/me-no-dev/ESPAsyncWebServer/issues/27), and seems resolved, but did not help. Obviously it's related to SD_MMC mounting by default at /sdcard (which I want). Can I configure `serveStatic(....)` an alternate way to be the desired behavior?

atanisoft, 2019-09-09:
```serveStatic("/", SD_MMC, "/");``` will do what you are after. The first parameter is the http endpoint, second is the VFS adapter and third is the path relative to the VFS adapter (internally /sdcard is prepended to the path and you shouldn't need to worry about that).

The reason you are seeing "/sdcard/sdcard/index.htm" is due to you passing in "/sdcard/" as the third parameter.

brettbeeson, 2019-09-09:
Thanks! I've got that working. Obviously over-thinking the whole SD_MMC mounting thing.

closed 2019-09-09 by brettbeeson