me-no-dev/ESPAsyncWebServer #594: How to make a rest endpoint / recive json

thbl, 2019-09-13:
Hi

I'm new to esp can some one give an example, on how to make a rest endpoint and parse json.
i tried running the sample code 
```
#include "AsyncJson.h"
#include "ArduinoJson.h"

AsyncCallbackJsonWebHandler* handler = new AsyncCallbackJsonWebHandler("/rest/endpoint", [](AsyncWebServerRequest *request, JsonVariant &json) {
  JsonObject& jsonObj = json.as<JsonObject>();
  // ...
});
server.addHandler(handler);
```
But this just gives me a compile error
```
   JsonObject& jsonObj = json.as<JsonObject>();

                                            ^

exit status 1
invalid initialization of non-const reference of type 'ArduinoJson::JsonObject& {aka ArduinoJson6120_000001::ObjectRef&}' from an rvalue of type 'ArduinoJson6120_000001::enable_if<true, ArduinoJson6120_000001::ObjectRef>::type {aka ArduinoJson6120_000001::ObjectRef}'

```

thbl, 2019-09-16:
solved just need to replace
 JsonObject& jsonObj = json.as<JsonObject>();
by
 JsonObject jsonObj = json.as<JsonObject>();

more info
https://arduinojson.org/v6/doc/upgrade/

closed 2019-11-29 by stale[bot]

willfaulds, 2020-03-21:
I found the correct line was
`JsonObject jsonObj = json.as<JsonObject>();`