me-no-dev/ESPAsyncWebServer #609: Communication between multiple ESP32 running Async Web Server

AkhileshThorat, 2019-09-27:
Hello,
I just started tinkering around AsyncWebServer and I must say it's one of the masterpiece I have came across. Learned lots of new stuff while tinkering.
I was curious whether it's possible to communicate between 2 ESP32 running ESPASyncWebServer(hosting a webpage as well as some  JSON data).
I tried using HTTP client, but failed to do so. Is there any thing i need to know/consider before i try even harder?

akiosTerr, 2019-09-29:
perhaps one could use web sockets, however, I don't know how you can use it with this library. it would be nice to see an example, I couldn't really understand it from the documentation

AkhileshThorat, 2019-09-30:
> 
> 
> perhaps one could use web sockets, however, I don't know how you can use it with this library. it would be nice to see an example, I couldn't really understand it from the documentation

Consider it as a distributed network of esp32 over a network (LAN here), each device keeps a check on each other and exchanges data to and fro between them.
I'm doubtful about how much sockets esp32 can handle at a time.

akiosTerr, 2019-09-30:
well, as far as I am concerned, an MQTT protocol is the best way to create an ESP network
some people do it with a Raspberry-pi as a master/broker, but you must be able to do it with a regular computer

AkhileshThorat, 2019-09-30:
> 
> 
> well, as far as I am concerned, an MQTT protocol is the best way to create an ESP network
> some people do it with a Raspberry-pi as a master/broker, but you must be able to do it with a regular computer

That's the case when you're connected to internet.
what if you're not connected to the internet, you are on a private network and connected on different subnets/gateways. Here comes the real challenge. 
You have to discover each other and create a peer to peer network using LAN(Ethernet here)

AkhileshThorat, 2019-10-01:
Hi @me-no-dev ,
I went through websockets documentation of yours, but didn't found suitable method to implement websockets client(ESP32) requesting data from websocket server(ESP32-Async-web-server).
Example:
How do I request data from http://{ESP-IP}/ws hosted on different ESP32?

me-no-dev, 2019-10-02:
not with this library :) this is a server library and you are asking for a client

AkhileshThorat, 2019-10-02:
> 
> 
> not with this library :) this is a server library and you are asking for a client

I had a slight hint, but wasn't sure. 
I tried http client between two ESP32 (each ESP has a AysncWebServer) but ESP runs into fault and keeps on rebooting.
Then I tried WiFi Client, but then server crashes after some time and becomes unresponsive .(have to reboot again).
Tried WebSockets  Client by  [https://github.com/Links2004/arduinoWebSockets](url), but doesn't help out.

Is it wrong implementation method which I'm following or I need to know something more?
@me-no-dev  your feedback will be of great help

akiosTerr, 2019-10-02:
I found this ESP protocol which enables you to communicate between ESP's without using wi-fi
its called ESP-Now: https://www.espressif.com/en/products/software/esp-now/overview
it may be useful in your case

AkhileshThorat, 2019-10-02:
> 
> 
> I found this ESP protocol which enables you to communicate between ESP's without using wi-fi
> its called ESP-Now: https://www.espressif.com/en/products/software/esp-now/overview
> it may be useful in your case

@akiosTerr  Thankyou for suggesting the solution, but it is a wireless communication protocol.
I'm looking for peer to peer connectivity between ESP32 through Ethernet(wired network).

zekageri, 2019-10-25:
I'am currently exchange lots of datas beetween my esp32s at my house. I have made a "Smart House" like project at home with lots of esps. I have a "server" esp at my room close to the router ( not neccessary to be close to it just in case) and a few more in the kitchen, in the balcony , at the stairs etc. They are communicating with each other on my local wifi network. They are all connected to the same network via wifi, and change datas on async websocket , regular arduino websocket and http protocols. Its easy as that. On my "server" esp there is a big bootstrap4 webpage all files served from its flash. When you investigate this webpage , you get a smart house controll panel, and if it is done loading , it is estabilishing websocket connections with all the esps in the house on its javascript side. When i hit a button or for example i want to turn on the mood lighting on the balcony if there are a bunch of friends, i hit that button , or turn on some color for it, the server immidiately send a websocket packet to the specific esp ( in this case to the balcony esp ) and its starting to pwm the leds. Both async websocket , regular websocket and http works fine for me. 
What do you want to implement?

AkhileshThorat, 2019-10-25:
Hi @zekageri ,
I have a similar scenario where I have several ESP's running web server. Each device  reads data from machines/meters and exchange data with each other(with other ESP on same network). Each device has it's own portal/web UI which displays the data monitored, device status and peer to peer connectivity status.
 I tried simple http client but ESP32 reboots repeatedly showing Guru Meditation error. As of now I'm using WiFi Client to exchange data between ESP's . and each ESP is serving a web page for showing /configuring data.

Talking about websockets, I was searching for client library which would work with async websocket present in Async Webserver Library. I'm investigating this matter, since I was tinkering around without diving deep into the matter.

zekageri, 2019-10-28:
Just use regular arduinowebsocket lib. 
It has a server and a client implementation too.
Btw in my case i separate the different datas with different special characters.
So technically if you send a packet from a websocket server to an other server, the other server will get the data, and if that data has a certain special character in its last char, then you know where it is from. I do it like that. :D

closed 2019-12-27 by AkhileshThorat

vburgosc, 2020-01-10:
> Hi @zekageri ,
> I have a similar scenario where I have several ESP's running web server. Each device reads data from machines/meters and exchange data with each other(with other ESP on same network). [...]

Hi! Did you figured out how to do it finally? I'm trying to do a similar thing, set up a HTTP webpage socket for client configuration and another socket for multiple ESP32 communication.

iChris93, 2020-06-19:
Any success with communication between two ESP32 running ESPAsyncWebServer? I’ve tried Websockets and could not seem to get that to work either.

gibo77, 2020-06-19:
Never tried ESP32. Mine is ESP8266. Client is Android React Native App NodeJS. Works like a rock!

vburgosc, 2020-06-19:
I did it, you have to initialize 2 TCP servers (with different ports), one for the web and the second for TCP transmisions between the ESP32 (it does work with ESP32 and ESP8266).

gsb, 2021-05-17:
vburgosc.
Did you or could you share your solution code? 
It would be nice for those of us looking for an example.
Thanks.