me-no-dev/ESPAsyncWebServer #613: binary project size at 1.2.3

reaper7, 2019-10-02:
today I updated ESPAsyncWebServer library to 1.2.3
and binary file for the espurna project increased its volume by about 100,000 bytes!

that's very strange because yesterday I was also updating the git version and everything was ok.

...so the same espurna project with ESPAsyncWebServer in the version from before today's changes:
```
IROM   : 458060          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28308   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1440  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 5540  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 27416 )         - zeroed variables      (global, static) in RAM/HEAP 

Użycie biblioteki ArduinoJson_5x w wersji 5.13.5 z folderu: C:\Programy\arduino_projekty\libraries\ArduinoJson_5x 
Użycie biblioteki ESP8266WiFi w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
Użycie biblioteki eeprom_rotate w wersji 0.9.2 z folderu: C:\Programy\arduino_projekty\libraries\eeprom_rotate 
Użycie biblioteki EEPROM w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\EEPROM 
Użycie biblioteki ESPAsyncTCP w wersji 1.2.2 z folderu: C:\Programy\arduino_projekty\libraries\ESPAsyncTCP 
Użycie biblioteki async-mqtt-client w wersji 0.8.2 z folderu: C:\Programy\arduino_projekty\libraries\async-mqtt-client 
Użycie biblioteki ArduinoOTA w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\ArduinoOTA 
Użycie biblioteki embedis w wersji 1.2.0 z folderu: C:\Programy\arduino_projekty\libraries\embedis 
Użycie biblioteki ESPAsyncWebServer w wersji 1.2.2 z folderu: C:\Programy\arduino_projekty\libraries\ESPAsyncWebServer 
Użycie biblioteki Hash w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\Hash 
Użycie biblioteki justwifi w wersji 2.0.2 z folderu: C:\Programy\arduino_projekty\libraries\justwifi 
Użycie biblioteki debounceevent w wersji 2.0.5 z folderu: C:\Programy\arduino_projekty\libraries\debounceevent 
Użycie biblioteki ESP8266mDNS w wersji 1.2 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS 
Użycie biblioteki Ticker w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\Ticker 
Użycie biblioteki Time w wersji 1.5 z folderu: C:\Programy\arduino_projekty\libraries\Time 
Użycie biblioteki NtpClient w wersji 2.5.0 z folderu: C:\Programy\arduino_projekty\libraries\NtpClient 
Użycie biblioteki DNSServer w wersji 1.1.1 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\DNSServer 
```

and with 1.2.3:
```
IROM   : 562580          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28312   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1644  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 7600  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 31408 )         - zeroed variables      (global, static) in RAM/HEAP

Użycie biblioteki ArduinoJson_5x w wersji 5.13.5 z folderu: C:\Programy\arduino_projekty\libraries\ArduinoJson_5x 
Użycie biblioteki ESP8266WiFi w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
Użycie biblioteki eeprom_rotate w wersji 0.9.2 z folderu: C:\Programy\arduino_projekty\libraries\eeprom_rotate 
Użycie biblioteki EEPROM w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\EEPROM 
Użycie biblioteki ESPAsyncTCP w wersji 1.2.2 z folderu: C:\Programy\arduino_projekty\libraries\ESPAsyncTCP 
Użycie biblioteki async-mqtt-client w wersji 0.8.2 z folderu: C:\Programy\arduino_projekty\libraries\async-mqtt-client 
Użycie biblioteki ArduinoOTA w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\ArduinoOTA 
Użycie biblioteki embedis w wersji 1.2.0 z folderu: C:\Programy\arduino_projekty\libraries\embedis 
Użycie biblioteki ESPAsyncWebServer w wersji 1.2.3 z folderu: C:\Programy\arduino_projekty\libraries\ESPAsyncWebServer 
Użycie biblioteki Hash w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\Hash 
Użycie biblioteki justwifi w wersji 2.0.2 z folderu: C:\Programy\arduino_projekty\libraries\justwifi 
Użycie biblioteki debounceevent w wersji 2.0.5 z folderu: C:\Programy\arduino_projekty\libraries\debounceevent 
Użycie biblioteki ESP8266mDNS w wersji 1.2 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS 
Użycie biblioteki Ticker w wersji 1.0 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\Ticker 
Użycie biblioteki Time w wersji 1.5 z folderu: C:\Programy\arduino_projekty\libraries\Time 
Użycie biblioteki NtpClient w wersji 2.5.0 z folderu: C:\Programy\arduino_projekty\libraries\NtpClient 
Użycie biblioteki DNSServer w wersji 1.1.1 z folderu: C:\Programy\arduino\hardware\esp8266com\esp8266\libraries\DNSServer 
```

me-no-dev, 2019-10-03:
this is rather strange... there is nothing in the code that would cost that. it's just fixes, no new features or anything.

me-no-dev, 2019-10-03:
one idea only... but not for 100K surely... see if you can revert this commit and see changes: https://github.com/me-no-dev/ESPAsyncWebServer/commit/6a4e2f532912f229ed8cb4699cd5114ba0b946c3

reaper7, 2019-10-03:
simple server example 
current 1.2.3:
```
Executable segment sizes:

IROM   : 364812          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27416   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1504  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 4556  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 29264 )         - zeroed variables      (global, static) in RAM/HEAP 

Szkic używa 398288 bajtów (38%) pamięci programu. Maksimum to 1023984 bajtów.
Zmienne globalne używają 35324 bajtów (43%) pamięci dynamicznej, pozostawiając 46596 bajtów dla zmiennych lokalnych. Maksimum to 81920 bajtów.
```

and revert to https://github.com/me-no-dev/ESPAsyncWebServer/commit/403752a942f797d735539606a904fefc312e91eb (one commit before _Support route parameters with regex patterns_):
```
Executable segment sizes:

IROM   : 257532          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27036   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1284  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 2448  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25272 )         - zeroed variables      (global, static) in RAM/HEAP 

Szkic używa 288300 bajtów (28%) pamięci programu. Maksimum to 1023984 bajtów.
Zmienne globalne używają 29004 bajtów (35%) pamięci dynamicznej, pozostawiając 52916 bajtów dla zmiennych lokalnych. Maksimum to 81920 bajtów.
```

me-no-dev, 2019-10-03:
so it was that commit that caused it? hmmm...

atanisoft, 2019-10-03:
@me-no-dev Does std::regex pull in exceptions?

reaper7, 2019-10-03:
probably...
I checked it on two different computers (but with the same environment)

reaper7, 2019-10-03:
@atanisoft - you're right,
if I comment part of WebHandlerImpl.h -> virtual bool canHandle

```
    virtual bool canHandle(AsyncWebServerRequest *request) override final{

      if(!_onRequest)
        return false;

      if(!(_method & request->method()))
        return false;

      if (_isRegex) {
#if 0
        std::regex rgx(_uri.c_str());
        std::smatch matches;
        std::string s(request->url().c_str());
        if(std::regex_search(s, matches, rgx)) {
          for (size_t i = 1; i < matches.size(); ++i) { // start from 1
            request->_addPathParam(matches[i].str().c_str());
          }
        } else {
          return false;
        }
#else
        return false;
#endif
      } else if (_uri.length() && _uri.endsWith("*")) {
        String uriTemplate = String(_uri);
	uriTemplate = uriTemplate.substring(0, uriTemplate.length() - 1);
        if (!request->url().startsWith(uriTemplate))
          return false;
      }
      else if(_uri.length() && (_uri != request->url() && !request->url().startsWith(_uri+"/")))
        return false;

      request->addInterestingHeader("ANY");
      return true;
    }
```
the file size is reduced from 398288 to 288704

atanisoft, 2019-10-03:
Hmm, I wonder if there is a lighter weight approach to this than std::regex. I know I was considering pulling this feature into my code but if the cost is 100k it is not an option for me.

reaper7, 2019-10-03:
for many larger projects this will be a problem

atanisoft, 2019-10-03:
Are there any smaller alternatives to std::regex that can be dropped in?

Also the current code can be improved by caching the std::regex object which will improve performance.

reaper7, 2019-10-15:
PR https://github.com/me-no-dev/ESPAsyncWebServer/pull/617 solved the problem

closed 2019-10-15 by reaper7