me-no-dev/ESPAsyncWebServer #616: Please do not have the Stale bot close unresolved issues

probonopd, 2019-10-05:
Please do not have the Stale bot close unresolved issues, thanks.

Just time passing does not magically make them solved or irrelevant.

me-no-dev, 2019-10-05:
well... having a whole lot or irrelevant issues also does not work. Nudging the issues actually raises the chance that they get resolved ;)

closed 2019-10-05 by me-no-dev

probonopd, 2019-10-05:
The bot could be configured to add a "needs-attention" label, for example. But closing helps no one, just frustrates the people who took the time to report an issue.

me-no-dev, 2019-10-05:
if anyone would have posted a message when the bot labeled the issue, it would have unlabeled it, then that email goes to separate inbox of mine

JamesNewton, 2019-10-05:
I'm just overjoyed that me-no-dev is actually looking at this repo. I thought it was abandoned. Apparently the Arduino IDE is no longer supported and I'm trying to convert to Platform.io so I can continue using this library. As to the repo, I don't care how you run it, just so long as you run it.

me-no-dev, 2019-10-06:
why Arduino IDE is not supported? works fine on my end. the repo is tested with Arduino IDE and PlatformIO.

probonopd, 2019-10-06:
Yes, kudos to @me-no-dev - you are doing fantastic work.

JamesNewton, 2019-10-07:
> why Arduino IDE is not supported?
There is absolutely no mention of supporting the Arduino IDE in the readme. It exclusively refers to Platform.IO. Any mention of the Arduino IDE was removed months or a year ago, I noticed when it was taken out but didn't comment on in then.

me-no-dev, 2019-10-07:
you drop it in libraries and it works. did not realise that it requires mention :)

JamesNewton, 2019-10-07:
It's not so much that it needs mention, it's that it /was/ mentioned and the mention was suddenly removed. And I was having serious issues with a new installation on a 2nd laptop (apparently unrelated) which made me think your support of the standard IDE was related (I found other issues later on) and so I assumed I needed to move to Platform.IO. 

BTW: I'm currently moving to Platform.IO because M$ have taken me to the dark side via the beguiling VSCode, and I'm having all sorts of fun trying to get ESP32 debugging working. 
https://community.platformio.org/t/esp32-pio-unified-debugger/4541/93
Hopefully the advice given will work despite one port seemingly disappearing. Very complex setup. And still slow as snot to compile. Makes the dev loop agonizing. But hopefully being able to debug will help. 

And you being back in the mix is a pure joy! Thanks.