me-no-dev/ESPAsyncWebServer #619: ESP32 is rebooting when trying to config ethernet interface

aliyehia92, 2019-10-10:
Hello.

I've been trying to solve this issue for 5 days now. I have an ESP32 hooked up to an ethernet PHY (LAN8720), and what I'm trying to do is allow users to config the ethernet interface from the web browser.

Basically I have a form containing 3 text inputs for the IP, subnet mask, and gateway, and when the user submits the form an XMLHttprequest is sent:

```
function sendSettings(inter) {

            var xhttp = new XMLHttpRequest();

            //wait for reply from server
            xhttp.onreadystatechange = function() {
                if (this.readyState == 4 && this.status == 200) {
                    var receivedString = this.responseText;
                    if (receivedString === "ok") {
                        alert("success");
                    }
                }
            };

            //get text input values
            var ip = document.getElementById("ipaddress").value;
            var subnet = document.getElementById("subnetmask").value;
            var gateway = document.getElementById("defgateway").value;

            xhttp.open("POST", "eth_config", true);
            xhttp.setRequestHeader("Content-Type", "application/x-www-form-urlencoded");
            xhttp.send("ip=" + ip + "&subnet=" + subnet + "&gateway=" + gateway);

            return false;
        }

```

On the ESP, the below code runs when the request is received:

```
server.on(file_name, HTTP_POST,[](AsyncWebServerRequest * request) {
            config_ethernet = true;
            request->send(200,"text/plain","ok");
}
```

the variable config_ethernet is there just to trigger an action on the main loop, I don't want to run any action on the asyncwebserver which can cause delays as per the documentation:

```
void loop(){
           if (config_ethernet) {
                      ETH.config(IPAddress(192, 168, 16, 105),
                      IPAddress(255, 255, 255, 0), IPAddress(192, 168, 16, 1));
                      config_ethernet = false;
           }
}
```

The reason I'm using static values for the config in the main loop is to make sure that the issue is not caused somehow by receiving the values from the browser then converting them to IPAddress objects. So I decided to just put in static values and run the test again and still the ESP reboots when I submit the form. Sometimes it works, sometimes it reboots on the second try, sometimes it reboots on the third try, so there's no pattern here.

Below is what I'm getting on the serial monitor:
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 25 - MAX 
assertion "new_rcv_ann_wnd <= 0xf[D][WiFiGeneric.cpp:417] _eventCallback(): ETH IP: 192.168.16.101, MASK: 192.168.16.1, GW: 255.255.255.0
Aff,f" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c", line 779, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x401179c7 on core 0 
 
Backtrace: 0x4008c618:0x3ffb3f80 0x4008c849:0x3ffb3fa0 0x401179c7:0x3ffb3fc0 0x40132fca:0x3ffb3ff0 0x40133059:0x3ffb4010 0x400e197a:0x3ffb4030 0x4012e488:0x3ffb4050 0x40088d05:0x3ffb4080 
 
Rebooting... 
```

What am I doing wrong here?

aliyehia92, 2019-10-10:
I just realized that I swapped the subnet and gateway in the main loop. Regardless, the issue is still there even after I fixed it.

BlueAndi, 2019-10-24:
Decode backtrace, which will give your more information.

aliyehia92, 2019-11-07:
anybody out there able to help? still haven't figured it out :/

closed 2020-01-20 by stale[bot]

CircuitSetup, 2020-07-21:
@BlueAndi did you ever figure out the issue you were having with this?

BlueAndi, 2020-07-21:
@CircuitSetup I didn't have this issue, the origin author was @aliyehia92. I requested to decode the backtrace, but never happened.

liamcharmer, 2024-11-03:
@aliyehia92 did you get this working?