me-no-dev/ESPAsyncWebServer #622: Specifying Date-Modified header for .h files in flash memory.

CelliesProjects, 2019-10-16:
Is it possible to set a 304 handler on `.h` files the same way as files served from spiffs of FFat?

I now use a construct something like:
``` 
...
  server.on( "/", HTTP_GET, [] ( AsyncWebServerRequest * request ) {
    if ( htmlUnmodified( request, date ) ) return request->send(304);
    AsyncWebServerResponse *response = request->beginResponse_P( 200, HEADER_HTML, index_htm, index_htm_len );
    response->addHeader( HEADER_LASTMODIFIED, date );
    request->send( response );
  });
...
static inline __attribute__((always_inline)) bool htmlUnmodified( const AsyncWebServerRequest * request, const char * date ) {
  return request->hasHeader( HEADER_MODIFIED_SINCE ) && request->header( HEADER_MODIFIED_SINCE ).equals( date );
}
````
This takes about 300ms.. 
Can this be done better?

me-no-dev, 2019-10-16:
esp32? WiFi.setSleep(false)?

CelliesProjects, 2019-10-16:
Indeed esp32.
WiFi.setsleep()?

me-no-dev, 2019-10-17:
yup. be default the wifi sleeps between beacons, that means that you send a packet, but the esp will not get it till the next beacon comes. That slows things down :) set the sleep to false and see if it improves

closed 2019-10-17 by CelliesProjects