me-no-dev/ESPAsyncWebServer #623: Use delay in Async Webserver

connectprabhjeet, 2019-10-16:
I'm using AsyncWebsockets of this library. How can I use delay between two tasks when server (NODEMCU) receives as text event from client (Browser).

For Example:
(*On text received from client [Javascript]) : if text == "LED" Turn on 4 LEDs and after 5000ms turn off all LEDs

I tried it but not getting worked
Thanks in advance.

proddy, 2019-10-16:
use a timer function, like [Ticker](https://github.com/esp8266/Arduino/tree/master/libraries/Ticker) ?

aliyehia92, 2019-10-17:
use [setInterval()](https://www.w3schools.com/jsref/met_win_setinterval.asp) and send an XMLHttpRequest every 5000ms. Something like:
```
setInterval(function () {
        var xhttp = new XMLHttpRequest();
        xhttp.onreadystatechange = function () {
            if (this.readyState == 4 && this.status == 200) {
                var receivedString = this.responseText;
                if (receivedString === "ok") {
                    console.log("success");
                }
            }
        };
        xhttp.open("GET", "turn-off", true);
        xhttp.send();
    }, 5000);
```
 That's on the browser side.

On the server side, you then do :
```
server.on("/turn-off", HTTP_GET, [](AsyncWebServerRequest * request) {

        //CODE TO TURN LEDS OFF

        request->send(200, "text/plain", "ok");
       }
);
```

closed 2019-12-30 by stale[bot]

jp-droid, 2021-02-25:
Can you still help me out with this?
ticker is not working

ozdemirsalih, 2024-10-07:
I know it's been a long time, but I wanted add my two cents for anyone wondering. DO NOT! use delay() or yield() with the ESPAsyncWebServer or ESPAsyncTCP, never! You actually don't need any timer. Just use millis() or micros(), both work fine. I didn't test the "polledtimeout", so I can't comment on that. But here's the basic millis() implementation in your loop;

`unsigned it milliscount;` 
`bool milliscountrecorded=false;`
`void setup() {`
`    //Setup your server here.`
`}`
`void loop() {`
`    if (milliscountrecorded) {`
`        if (millis()-milliscount>999) {`
`            milliscountrecorded=false;`
`            //The code here will be executed once in every second.`
`        }`
`    }`
`    else {`
`        milliscount=millis();`
`        milliscountrecorded=true;`
`    }`
`}`

This approach takes minimal amount of clock cycles for both ARM and RISC-V chips. The "polledtimeout" might be a better implementation, I really don't know, but if you certainly want anything better than this for the clock cycles, you need to look into xTensa Assembly. I hope it helped.