me-no-dev/ESPAsyncWebServer #624: Q: If there any way to deliver out a webpage bigger than the free Heap

JoanMCD, 2019-10-18:
Hello, I walk through the documentation and other issues searching a way to send out (for now) a webpage with 137k with a free heap of only 25k.

My hope was the streaming, but – as I understand – also here all data will be collected first before sending. 
I also checked the chunked encoding. 
All the flowing functions work with small data.

**Normal:**
```
  server.on("/page.cgi", HTTP_GET,  [](AsyncWebServerRequest * request) {
    request->send(200, "text/html", getData (all););
  });

```
**Stream:**
```
server.on("/page.cgi", HTTP_GET, [](AsyncWebServerRequest * request) {
      AsyncResponseStream  *response = request->beginResponseStream("text/html");
      response->print(getData (first_part));
      response->print(getData (second_part);
      request->send(response);
    });
```

**Chunked encoding:**
```
server.on("/page.cgi", HTTP_GET, [](AsyncWebServerRequest * request) {
      AsyncWebServerResponse  *response = request->beginChunkedResponse("text/html");
      response->sendChunked(getData (first_part));
      response->sendChunked(getData (fsecond_part));
      request->send(response);
    });
```

Of course, I can reload the content by Ajax or Websocket or can create sub pages – but that’s not a nice way come fast to the goal.

Blend3rman, 2019-10-19:
Unfortunately, the library (in fact TCP protocol itself) only supports sending one response to one request. Once `request->send()` is called, the connection is closed, making it impossible to add extra content after the fact. You're better off sticking with AJAX instead of trying to ram a large file into heap.

JoanMCD, 2019-10-19:
OK, thank you for you infos. 
Why it's not possible to send out multiple fragments before send the final connection close. 
It can not only a problem of TCP because it's working elsewhere. So the problem is to handover the full page to one TCP call. And the AsyncTCP  is from the same programmer. Over all - maybe it's clear for others - it would be helpful to add, that **_the page must smaller then the free heap (see headline)_**  in bold red into the docu :)

bot1131357, 2019-12-06:
Might I ask, why can't you use Chunked Response? You can break data down and send them in small chunks without overloading the heap memory.

sfranzyshen, 2020-02-06:
stale bot is not useful ... it degrades development ... please eliminate this feature from this repository ...

closed 2020-04-20 by stale[bot]