me-no-dev/ESPAsyncWebServer #628: Web pages load very slow. More than 1 second for a 310 B file.

GeorgeFlorian, 2019-10-21:
**TL;DR:** I am currently having to wait a very long time for a page to load.

I've booted up an ESP32 that has a web server on it and I am experiencing very long loading times. This is a new issue, because the pages were loading in under 1 second.

[image]

I know that a simple solution would be to not serve the js file locally, but that isn't the issue, because, as you can see `networks_placeholders.html` has a size of 310 B and it still takes a very long time to load it.

Here is the page:

```
    server.on("/IP-Config", HTTP_GET, [](AsyncWebServerRequest *request){
      if(userFlag) {
        if(!request->authenticate(value_login[0].c_str(), value_login[1].c_str()))
          return request->requestAuthentication(NULL,false);            
        request->send(SPIFFS, "/AP_configPage.html", "text/html", false, processor);
      } else {
        request->send(SPIFFS, "/AP_configPage.html", "text/html", false, processor);
      }  
    });
```
```
<!DOCTYPE html>
<html>    
<head>
    <meta charset = "utf-8">
    <meta name = "viewport" content = "width=device-width, initial-scale=1.0">
    <!-- Chrome, Firefox OS and Opera -->
    <link rel="icon" type="image/png" sizes="32x32" href="/favicon-32x32.png">
    <link rel="icon" type="image/png" sizes="16x16" href="/favicon-16x16.png">
    <link rel="shortcut icon" href="/favicon.ico">
    <!-- Tab Color Android Chrome -->
    <meta name="theme-color" content="#e11422">
    
    <link rel = "stylesheet" type = "text/css" href = "newMaster.css">   
    <title>IP Config</title>
    <script type="text/javascript" src="jquery-1.12.4.min.js"></script>
</head>
    
<body>
    <div class="logo_container">
        <button class="back_button" onclick="goBack()">Go Back</button>
        <img url="/logo.png">
        <span class = "version">PPD Display Controller</br>Version: %PH_Version%</span>
    </div>
    <div class="top_box">
            <div class = "box_head">
                    <div class = "title"><h1>Available Networks</h1></div>
                </div>
        <div class = "text_box">%PLACEHOLDER_NETWORKS%</div>
    </div>

    <div class="center_box">
        <div class = "box_head">
            <div class = "title"><h1>Setup - Network</h1></div>
        </div>
        <form method = "post" action="/dhcpIP">
            <input class = "button" type="submit" name="dhcp" value="DHCP IP" title="Redirect to DHCP IP Config page">
            <input class = "button" type="submit" name="static" value="Static IP" title="Redirect to Static IP Config page" formaction="/staticIP">
        </form>
    </div>    
</body>  
<script>
    $(document).ready( function(){
    $('.text_box').load('networks_placeholders.html');
    refresh();
    });
    
    function refresh(){
        setTimeout( function() {
            $('.text_box').load('networks_placeholders.html');
            refresh();
        }, 10000);
    }
</script>
<script>
    function goBack() {
    window.history.back();
    }
</script>
</html>
```

me-no-dev, 2019-10-22:
try WiFi.setSleep(false) after WiFi.connect(); :)

proddy, 2019-10-22:
WiFi.begin(); ;)

GeorgeFlorian, 2019-10-23:
> try WiFi.setSleep(false) after WiFi.connect(); :)

I've encountered a new issue: the server doesn't work in AP anymore. My webpages load a very long time until I get `The site can't be reached` and I am also continuously getting:
```
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[W][AsyncTCP.cpp:962] _poll(): ack timeout 4
```

The thing is that it worked. I have changed nothing in the code. I've just updated the libraries and nothing works right now.

GeorgeFlorian, 2019-10-23:
I suspect that the ESP32 was, in actuality, too far away from the AP and that it barely had any WiFi signal. Moving it closer to the AP improved the loading times and fixed my issue.

> try WiFi.setSleep(false) after WiFi.begin(); :)

I have added this in the code, but I don't know if it makes any difference. I will leave it like this, because for my project the WiFi must always be up.

closed 2019-10-23 by GeorgeFlorian