me-no-dev/ESPAsyncWebServer #630: Webserver with Ethernet + AP mode.

zekageri, 2019-10-25:
Hello guys!

### I have a very very big problem with accessing the esp32's webserver from AP mode.
**I'am currently using async server with Ethernet library.**

_When i investigating the webpage from my local network with pc or laptop throught its Ethernet interface, the webpage is very very very fast and stable.
If i turn the AP mode on, doesnt matter where in the program, sometimes i get very very low wifi signal, and even when the signal is good, the esp32 is restarting with various error messages or if it is not restarting immidiately the webpage does not load._

**Maybe the problem is that i want to use its Ap mode when i already using the Ethernet?**

**I tried so many ways to turn the AP mode on.**
 _Turning just WiFi.softAP("SoftAP"); without password or anything.
 Tried with password, tried with channels and other settings, tried with static IP
 Tried to disconnect the wifi first, and then turnin on AP mode.
 Tried disconnect and WiFi.enableAP(true); first, tried separately.
 etc..._

Error messages are mostly idle cpu1.
I'am running an i2s adc read on the cpu1 but the problem was not that because i commented it out and the problem presist.
Sometimes it is restarting with heap poisoning. 
But the problem occours just when i try to reach the webpage from AP mode with mobile phone.
Never from local network trought Ethernet.

**My loop is looks like this:**
```
// Two task continously running for the i2s adc
    xTaskCreatePinnedToCore ( loop0, "v_getIMU0", 3000, NULL, 0, &TaskHandle_1, 1 );
    xTaskCreatePinnedToCore ( loop1, "v_getIMU1", 5000, NULL, 0, &TaskHandle_2, 1 );

static void loop0(void * pvParameters){
  for( ;; ){    
        Sampling();             // RETRIVE DATA FROM DMA BUFFER
  }
}

static void loop1(void * pvParameters){
for( ;; ){
    if(!Ota_is_Started){
    Send_Amplitude(Min_Max_Amp(i2s_read_buff)); // RETRIVE DATA FROM i2s_read_buff FOR CALCULATION AND VISUALIZATION
    Send_Full_Buff_or_Min_Max();                          // Just checking a boolean variable
  }
  }
}

static const inline void Main_Loop(){
    Tag_Timer_Enabler_in_Sec(2);            // Just checking a boolean variable
    ArduinoOTA.handle();                          // OVER THE AIR* PROGRAM UPLOAD HANDLE
    webSocket.loop();                             // WEBSOCKET PACKET LOOP
    Handle_Tag_Reading();                    // Just checking a boolean variable
    Refresh_Time_in_Sec(1);                   // Running every 1 sec
    Temp_to_Web();                              // Just checking a boolean variable
    Set_Time_From_Web();                    // Just checking a boolean variable
}
```

GeorgeFlorian, 2019-10-28:
> Error messages are mostly idle cpu1.
Try adding `delay(2);` at the end of `void loop() {}`.

> Maybe the problem is that i want to use its Ap mode when i already using the Ethernet?

What do you mean by this ?

I think that starting ETH is like starting WiFi STA_MODE. I don't start ETH if I plan to use AP_MODE.

You can use this to set WiFi to AP_MODE and to check if it actually went in AP_MODE:
```
Serial.println(WiFi.mode(WIFI_AP) ? "Controller went in AP Mode !" : "ERROR: Controller couldn't go in AP_MODE. AP_STATION_MODE will start.");
```

I separate my code into two blocks: AP and STA and I let the ESP be in only one state at a time.
`Start only AP -> Use AP -> restart ESP -> Start only STA -> Use STA`

zekageri, 2019-10-28:
### Problem Solved.
The problem was that i wanted to serve many requests when the user investigated the webpage.
**I was created the page on bootstrap template and i had to load a shit ton of libs to the web.
The web requested my js file , bootstrap js file , chart js file , zoom js file for chart js, jquery js file several css files and some png files from flash.
Now i merged all the js file into one js file and all the css files to one. That way the web requests just three responses from the esp and i can gladly use the AP while i use the ethernet interface.**

closed 2019-10-28 by zekageri

ArminPP, 2020-12-25:
> **I'am currently using async server with Ethernet library.**
> 
> _When i investigating the webpage from my local network with pc or laptop throught its Ethernet interface, the webpage is very very very fast and stable._

Hello zekageri,
I cannot operate the library with an ethernet shield (W5500).
Could you please show me how you did it?

Thank you very much,
Armin

zekageri, 2020-12-25:
Sorry i'm not using a shield and i use CONFIG_PHY_LAN8720. 
I'm using a library for it and a little tweak in the header file. 
I get it from Me_No_Dev a while ago. If you still need my help i will show the library and etc. I don't know if it is very different or not from the W5500

ArminPP, 2020-12-26:
Hello zekageri,
It would be nice if you could share the tweak, maybe I can find a solution for the W5500.

Best regards,
Armin

kevinkerker, 2024-01-04:
@zekageri Could you please share the tweak / library you got from Me_No_Dev? I'm currently trying to set up the asyncwebserver with LAN8720.

zekageri, 2024-01-04:
I don't use neither asyncwebserver and the tweaked library.
I recommend for you to transition to [PsychicHttp](https://github.com/hoeken/PsychicHttp) because this is the successor of async tcp. The syntax is almost the same but without the bugs and crashes. As of the LAN8720 lib, it just works with the built in ETH library now.

This is how i use the ETH_LAN8720

```cpp
#include <ETH.h>

// This is my GPIO config
#define ETH8720_PHY_ADDR  1
#define PIN_PHY_POWER -1
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18

void beginETH(){
   WiFi.onEvent([this](WiFiEvent_t event, WiFiEventInfo_t info){
      if( event == ARDUINO_EVENT_ETH_GOT_IP ){
          Serial.printf("[Ethernet] - Got %s IP\n",ETH.localIP().toString().c_str());
      }else if( event == ARDUINO_EVENT_ETH_DISCONNECTED ){
          Serial.printf("[Ethernet] - Disconnected");
      }
   });
   ETH.begin(
        ETH8720_PHY_ADDR,
        PIN_PHY_POWER,
        PIN_SMI_MDC,
        PIN_SMI_MDIO,
        ETH_PHY_LAN8720,
        ETH_CLK_MODE
    );
}
```