me-no-dev/ESPAsyncWebServer #633: AsyncCallbackWebHandler support request for variables in URI

BlueAndi, 2019-10-27:
The webserver of the arduino-esp32 framework supports variables in the URI, e.g.
```
srv.on("/rest/api/v1/display/slot/{}/text", slotText);
```
The registered request handler will received the ```{}``` as an argument.

It would be good to have the same support in the ESPAsyncWebServer as well.

Bmooij, 2019-10-31:
The AsyncWebserver has already support for variables in the URI.

Check the docs [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/README.md#path-variable)  and for examples [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/examples/regex_patterns/regex_patterns.ino)

BlueAndi, 2019-10-31:
Introduced in the last version, yes, I saw it. The drawback is the necessary space for regex. Would it be possible without regex as alternative?

Bmooij, 2019-10-31:
@me-no-dev I would like to add support for this function, without making a mess of it. Can you please give me feedback on my plan?

I would like to proposes an `Uri` object for the uri. 

The usage of the `Uri` object would be as follows:

a static uri
```cpp
srv.on("/a/static/uri", callback);
srv.on(Uri("/an/other/static/uri"), callback);
```

a regex uri
```cpp
srv.on(UriRegex("^\\/an\\/regex\\/uri\\/[0-9]+$"), callback);
```

a bracket uri
```cpp
srv.on(UriBrackets("/an/uri/with/{}/backets"), callback);
```

The `Uri` class will look someting like this:
```cpp
class Uri {
    public:
          virtual bool canHandle(AsyncWebServerRequest *request);
    
    protected:
          void _addPathParam(AsyncWebServerRequest *request, const char* param);
}
```

This implementation will also get rid of the `ASYNCWEBSERVER_REGEX` flag.
By importing `uri/UriRegex.h` automatically 'regex' will be included.

BlueAndi, 2019-12-31:
@me-no-dev Your answer is missing. Solution would be available, see previous comments.

BlueAndi, 2020-02-29:
ping ;-)

BlueAndi, 2020-04-30:
@me-no-dev Please comment.

RomeHein, 2020-05-27:
This is too bad, because this feature is really missing. atm it's quite difficult to make a restful api. I was looking at the `addRewrite` function, but told myself that it should be included in the library without having to write painful parser or set build flags for regex.

BlueAndi, 2020-07-27:
Last try, before I give up.

closed 2020-10-10 by stale[bot]