me-no-dev/ESPAsyncWebServer #634: AsyncTCP, rebooting while file serving

d0ct0rd-dalimo, 2019-10-28:
Hi all!

When I use serveStatic(), it triggers watchdog timer and reboot device when I serve large file (~25kB).
JSON responses are handled correctly.

My serverUp code:

```void serverUp() {
  
  esp_wifi_set_ps (WIFI_PS_NONE);
  server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");  // Обработчик запроса домашней страницы
  server.serveStatic("/gz/css", SPIFFS, "/gz/css");
  server.serveStatic("/gz/js", SPIFFS, "/gz/js");
  server.serveStatic("/public", SPIFFS, "/logs");
  //server.serveStatic("/settings.json", SPIFFS, "/settings.json").setAuthentication(auth_username, device_key);
  server.on("/set_track_times", HTTP_POST, [](AsyncWebServerRequest *request) {
    saveTimeLimSets(request);
    req_millis = millis();
  });
  server.on("/file_list", HTTP_GET, [](AsyncWebServerRequest *request) {
    handleFileList(request);
    req_millis = millis();
  });
  server.on("/file_from_sd", HTTP_GET, [](AsyncWebServerRequest *request) {
    handleSDFileList(request);
    req_millis = millis();
  });
  server.on("/get_dev_data", [](AsyncWebServerRequest *request) {
    getDevData(dataObject.ts_now, request);
    req_millis = millis();
  });
  server.on("/get_tts", [](AsyncWebServerRequest *request) {
    request->send(403, "text/plain", "FORBIDDEN");
    req_millis = millis();
  });

  server.on("/op_status", [](AsyncWebServerRequest *request) {
    sendFinishCode(request);
    req_millis = millis();  
  });
  
  server.serveStatic("/config", SPIFFS, "/config.html");
  server.onNotFound([](AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
  }); //  Обработчик запроса несуществующей страницы

  //server.setRxTimeout(5000);
  //server.setAckTimeout(5000);
  //Serial.print("No delay: ");
//  Serial.println(server.getNoDelay() ? "true" : "false");
  server.begin();
}
```

Error:

```E (69609) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
14:32:42.019 -> E (69609) task_wdt:  - async_tcp (CPU 0/1)
14:32:42.019 -> E (69609) task_wdt: Tasks currently running:
14:32:42.019 -> E (69609) task_wdt: CPU 0: IDLE0
14:32:42.019 -> E (69609) task_wdt: CPU 1: IDLE1
14:32:42.019 -> E (69609) task_wdt: Aborting.
14:32:42.019 -> abort() was called at PC 0x400f499f on core 0
14:32:42.019 -> 
14:32:42.019 -> Backtrace: 0x4008d0c8:0x3ffbe170 0x4008d2f9:0x3ffbe190 0x400f499f:0x3ffbe1b0 0x400853d1:0x3ffbe1d0 0x40168557:0x3ffbc460 0x400f67e3:0x3ffbc480 0x4008afd1:0x3ffbc4a0 0x400897dd:0x3ffbc4c0
14:32:42.052 -> 
14:32:42.052 -> Rebooting...

```

I've tried to add #define CONFIG_ASYNC_TCP_RUNNING_CORE 1, but it made no sense

zekageri, 2019-11-04:
I realized that if you send a bunch of responses to the front end, the async server is going to crash itself.
I'am using the response functions like this:

```
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncWebServerResponse* response = request->beginResponse(SPIFFS, "/Index.html", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
});
```

And not servestatic. These files are gzipped without the "gz" extensions. This is required if you want to serve the files to IOS. IOS can't recognize gz extensions, however it can unpack them if the response header explicitly sais that this file is gzipped. Nvm. When i served a relatively small web (7kb) to the browser, and loaded so many libs to the front end like bootstrap js files, jquery js and some css and else.. the exp crashes. I realized that it is because the browser do these requests much more faster then the esp can handle it. So i put all my js files into one js file , and all my css files into one css file. Now i have 3 request from the browser and i serve a 100kb!!! js file and a 30kb css file and a 7kb index html file. The page load time is significantly faster like this and no more crashes.

proddy, 2019-11-04:
@zekageri even faster is obfuscating the files, gzipping and hard coding the content into PROGMEM which totally avoids using SPIFFS.

d0ct0rd-dalimo, 2019-11-08:
My project requires external library for plotting on a web-page and I should use AP WiFi mode, so I store it, js, css and html's on SPIFFS and some of them are gzipped

d0ct0rd-dalimo, 2019-11-08:
I combined several js into one large and several css into one and gzipped all of them. But I still use serveStatic, I'm trying it... There is no that issue yet

d0ct0rd-dalimo, 2019-11-11:
I've tested with combined files and I saw no issues

closed 2019-11-11 by d0ct0rd-dalimo