me-no-dev/ESPAsyncWebServer #635: ESP32 crash with multiple client connected in AP mode.

ttlappalainen, 2019-10-30:
Thanks for exellent library.

I have pages containing some style sheets, external scripts and images. Each part is max 10 kB and all are located on FLASH not SPIFFS. When the page has been loaded, it requests data from ESP32 WebServer with short GET and also gets short response of values. In running state all clients does value request in every 0.2 sec, but effective data to be transfered is <50 bytes. I also tested by reducing pages so that there is no images and so max. part size is about 4kB at start, but that did not seem to have any effect.

I have problem with AP mode. I have tested Async Web Server with AP_STA mode and it seem to work fine with max one client connected to AP. Yesterday I had 6 computers connected to ESP32 STA side for about 4 hours making about 400000 data requests. There was no problem to connect new computer, which causes naturally full page update with all images etc. On one computer I had connection to ESP32 AP and it was also running fine.

The problem is that when I connect new client to ESP32 AP, it start to crash. Sometimes it runs only 20 seconds, sometimes I have seen it running up to 20 min. Normally I got error "CORRUPT HEAP:..." Also with multiple client to AP device stops sometimes responding for 10 sec, but then continues without crash.

The code is simple. Just define mode AP_STA, start AP and connect STA. But I have the same problem also with just AP mode.

I use PlatformIO with Arduino framework. I have updated framework and your libraries to latest.

Any idea how to find cause?

me-no-dev, 2019-10-30:
You can't :) it's somewhere in the wifi lib that is closed source. AP is software realised, unlike STA which runs mostly hardware. It will be fixed hopefully, but no clue when or what it is.

ttlappalainen, 2019-10-30:
Have you seen similar problem with AP and multiple clients?

me-no-dev, 2019-10-30:
i've seen different issues with AP. You should try to have the least possible requests. Employ content caching so you do not have to serve all files all of the time. Other than that... no clue how to help :)

BlueAndi, 2019-10-31:
Corrupt heap sounds like a wild pointer. You could analyse the backtrace to see the last function calls if the stack was not overwritten.

ttlappalainen, 2019-10-31:
Sure corrupt heap means leaking and most likely wild pointer. But if it as me-no-dev responded at first that wifi lib is closed, there is no sense for me to spend any extra hours with that. I need to define product WiFi so that AP is only for defining station. Hopefully it will be fixed in future.

ttlappalainen, 2019-11-07:
The problem is on configuring AP IP and using something else than default. I have solution, which seem to work and wrote it on issue https://github.com/espressif/arduino-esp32/issues/2025#

closed 2020-01-20 by stale[bot]