me-no-dev/ESPAsyncWebServer #638: How cancel a POST / file upload operation

r2d2r01, 2019-10-30:
Hi, suite at a cancel operation inside handler below (http_handle_dispupdate_upload), I send an answer (below) to client to interrupt process, but packets continues to comes until client processes 404 response sended. Is exist a way to broke link more instantaneously and avoid packet received ?
Thank's
Frederic

**Cancel response:**
`sprintf (_buffer,"%sCalculation in progress, must interrupt upload",ph);
answ = request->beginResponse(400, "text/plain", _buffer);
answ->addHeader("Connection", "close");
answ->addHeader("Access-Control-Allow-Origin", "*");
request->send(ans);
`
**Update handler:**
`webserver.on("/autoupd", HTTP_POST, http_handle_autoupdate_request, http_handle_dispupdate_upload);
`

**ESP32 traces / packets continue (1436..64100bytes)**
app_main ------------------------------------------------ 8
20191030 20:06:50 RAM:(U=329/F=226/Mfsb=224)Kb
AUTO-UPD: Upload 1436 bytes (1436)
AUTO-UPD: Upload 1436 bytes (2872)
.....
AUTO-UPD: Upload 1436 bytes (61748)
AUTO-UPD: Upload 1436 bytes (63184)
AUTO-UPD: Upload 916 bytes (64100)
app_main ------------------------------------------------ 9
20191030 20:06:55 RAM:(U=329/F=224/Mfsb=213)Kb

**Cmd sent & traces**
`$ curl -v -F "file=@fw.bin" -H "X-File-Size: 1928400" 192.168.0.150/autoupd --trace-time
20:05:43.442301 * STATE: INIT => CONNECT handle 0x200483f8; line 1429 (connection #-5000)
20:05:43.444233 * Added connection 0. The cache now contains 1 members
20:05:43.445304 *   Trying 192.168.0.150...
20:05:43.445481 * TCP_NODELAY set
20:05:43.445958 * STATE: CONNECT => WAITCONNECT handle 0x200483f8; line 1481 (connection #0)
20:05:43.650520 * Connected to 192.168.0.150 (192.168.0.150) port 80 (#0)
20:05:43.651107 * STATE: WAITCONNECT => SENDPROTOCONNECT handle 0x200483f8; line 1598 (connection #0)
20:05:43.651663 * Marked for [keep alive]: HTTP default
20:05:43.652280 * STATE: SENDPROTOCONNECT => DO handle 0x200483f8; line 1616 (connection #0)
20:05:43.653218 > POST /autoupd HTTP/1.1
20:05:43.653218 > Host: 192.168.0.150
20:05:43.653218 > User-Agent: curl/7.61.0
20:05:43.653218 > Accept: */*
20:05:43.653218 > X-File-Size: 1928400
20:05:43.653218 > Content-Length: 1928604
20:05:43.653218 > Content-Type: multipart/form-data; boundary=------------------------832cd4d415d3bb9c
20:05:43.653218 > Expect: 100-continue
20:05:43.653218 >
20:05:43.658010 * STATE: DO => DO_DONE handle 0x200483f8; line 1695 (connection #0)
20:05:43.658549 * STATE: DO_DONE => WAITPERFORM handle 0x200483f8; line 1822 (connection #0)
20:05:43.659083 * STATE: WAITPERFORM => PERFORM handle 0x200483f8; line 1837 (connection #0)
20:05:43.664584 * HTTP 1.1 or later with persistent connection, pipelining supported
20:05:43.665071 < HTTP/1.1 100 Continue
20:05:43.690986 * HTTP 1.1 or later with persistent connection, pipelining supported
20:05:43.691411 < HTTP/1.1 400 Bad Request
20:05:43.691821 < Content-Length: 51
20:05:43.692202 < Content-Type: text/plain
20:05:43.692582 * Marked for [closure]: Connection: close used
20:05:43.692959 < Connection: close
20:05:43.693341 < Connection: close
20:05:43.693724 < Access-Control-Allow-Origin: *
20:05:43.694104 < Accept-Ranges: none
20:05:43.694490 <

20:05:43.695085 * we are done reading and this is set to close, stop send
20:05:43.695490 * STATE: PERFORM => DONE handle 0x200483f8; line 2008 (connection #0)
20:05:43.695869 * multi_done
20:05:43.696293 * Closing connection 0
20:05:43.696672 * The cache now contains 0 members
20:05:43.697190 * Expire cleared
**AUTO-UPD: Calculation in progress, must interrupt upload**
`

BlueAndi, 2020-01-12:
Did you try to send ```request->send(413, "text/plain", "Upload aborted.");```?
If the client receives the 413, it should stop sending further data.
But to be honest, I did not test it yet.

r2d2r01, 2020-01-13:
Hello BlueAndi, I try a lot of break/error/...msg ... but none provide good reaction.
Thank's for answer

closed 2020-03-27 by stale[bot]