me-no-dev/ESPAsyncWebServer #640: What happened to my second core?! - Readme needs updating

rudolf81, 2019-11-02:
Firstly - great lib guys! - very fine work.

I ran into [this ERR 128 on my ESP32](https://github.com/espressif/arduino-esp32/issues/1921) with the standard WebServer class... Unable to figure it out, I followed one of the comments for AsyncWebServer, and - with some suspicion, I started porting my app over to AsyncWebServer.

It did indeed solved the issue and my web interface loaded fine, however I also have some code I need to run in the second core, and I noticed this is no longer working!

Seems AsyncTCP uses the second core...

AsyncWebServer is probably fine for most users porting Arduino code over to ESP32 but if you have an app written from the ground up to use both cores - losing functionality of one is a pretty steep price.

AsyncWebServer's readme file does warn to not use yield, etc - but no mention anywhere of the loss of the second core.

Don't you guys think this would be handy to know when considering adding AsyncWebServer to your project?

zekageri, 2019-11-04:
It's simple logic. The esp's documentation said that all the wifi and other functions running on the pro cpu  (0th core) and your code can run on the app cpu (1st core).

me21, 2019-11-04:
But he said his code ran on both cores before...

rudolf81, 2019-11-04:
On the ESP32 you can have concurrent execution on both cores - you run the main loop() method on one core, and you create a pinned task on the second core. This works, and you can read up about it / check it out on youtube.
However when you use ESPAsyncWebServer (which uses AsyncTCP) - it claims one of the cores, and you can no longer have concurrent execution.
This is fine, I just think the project author should maybe warn people about this.

atanisoft, 2019-11-04:
@rudolf81 where does AsyncTCP or EspAsyncWebServer claim a core? It may create a task that runs on a given core but it does not monopolize the core to non-usable for application code. When there is no data to send/receive any async task is idle and will not consume cpu time

me-no-dev, 2019-11-04:
https://github.com/me-no-dev/AsyncTCP/blob/master/src/AsyncTCP.h#L35

closed 2019-11-04 by me-no-dev