me-no-dev/ESPAsyncWebServer #647: esp32-cam example

array81, 2019-11-13:
Is possible have an example to use ESPAsyncWebServer to send esp32-cam streaming video to html webpage?
I jusr use ESPAsyncWebServer to create a webserver with html GUI on esp32-cam. From GUI I can control some relay. Now I'd like put on the same GUI also esp32-cam video but I cannot understand how can I do it.

me-no-dev, 2019-11-13:
i've already done it, but can not link it at this moment. will try to do over the weekend :)

array81, 2019-11-13:
Thanks. You would give me great pleasure.

me-no-dev, 2019-11-22:
https://gist.github.com/me-no-dev/d34fba51a8f059ac559bf62002e61aa3 :)

me-no-dev, 2019-11-22:
serving the index is up to you, this is the rest of the functionality :)

closed 2019-11-22 by me-no-dev

japhsc, 2022-02-07:
I had the same issue. 
I suspect line 136: 
`log_printf("Size: %uKB, Time: %ums (%.1ffps)\n", _jpg_buf_len/1024, fp);`

Try replacing it with:
`log_printf("Size: %uKB, Time: %.0fms (%.1ffps)\n", _jpg_buf_len/1024, fp, 1000./fp);
`