me-no-dev/ESPAsyncWebServer #648: Enabling regex causes crash and bootloop

teemue, 2019-11-13:
Using PlatformIO and latest ESPAsyncWebServer. 
Accessing http://ip/hello works but if I access /sensor, ESP crashes and bootloops.

  ```
server.on("/hello", HTTP_GET, [](AsyncWebServerRequest *request) {
    Serial.println("hello");
    request->send(200, "text / plain", "Other URL");
  });

  server.on("^\\/sensor\\/([0-9]+)$", HTTP_GET, [](AsyncWebServerRequest *request) {
    String sensorId = request->pathArg(0);
  });
```

platform.ini:
```
[env:esp01]
platform = espressif8266
board = esp01_1m
framework = arduino
build_flags = 
  -DASYNCWEBSERVER_REGEX
```

Bmooij, 2019-11-16:
Does it still crash when you add an response to the sensors request?

```
server.on("^\\/sensor\\/([0-9]+)$", HTTP_GET, [](AsyncWebServerRequest *request) {
  String sensorId = request->pathArg(0);
  request->send(200, "text / plain", "sensor: " + sensorId);
});
```

teemue, 2019-11-16:
It crashes. Here's the full code:
```
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h>

AsyncWebServer server(80);
DNSServer dns;
#define HOST_NAME "ESP"

void setup()
{
  Serial.begin(115200,SERIAL_8N1,SERIAL_TX_ONLY);
  AsyncWiFiManager wifiManager(&server, &dns);
  if (!wifiManager.autoConnect())
  {
    ESP.reset();
    delay(1000);
  }

  String hostNameWifi = HOST_NAME;
  hostNameWifi.concat(".local");
  WiFi.hostname(hostNameWifi);

  server.on("/hello", HTTP_GET, [](AsyncWebServerRequest *request) {
    Serial.println("hello");
    request->send(200, "text / plain", "Other URL");
  });

  server.on("^\\/sensor\\/([0-9]+)$", HTTP_GET, [](AsyncWebServerRequest *request) {
    String sensorId = request->pathArg(0);
    request->send(200, "text / plain", "sensor: " + sensorId);
  });

  server.begin();
}

void loop()
{
}
```

teemue, 2019-11-17:
I tried also without WifiManager (only ESP8266WiFi and  ESPAsyncWebServer included), no effect.

niklasdoerfler, 2019-12-23:
I can observe the same behaviour. Is there already a solution to the problem?

Bmooij, 2019-12-27:
There is an issue with regex and the esp8266. 
See: 
https://github.com/esp8266/Arduino/issues/6198 
https://github.com/esp8266/Arduino/issues/6952

There is no fix for now. When I have some time, I will rewrite the regex part and create an new pull request

closed 2020-03-11 by stale[bot]

kjyv, 2022-08-02:
still an issue, so should be reopened

phil31, 2023-03-29:
Hello users, 

2023, same problem !
build_flags = 
	-DCORE_DEBUG_LEVEL=5
	; -DASYNCWEBSERVER_REGEX=1

enable the DASYNCWEBSERVER_REGEX flag, then crash on requests !

any suggestion to use regex in url please ?
2023-03-29 : example "regex_patterns.ino" is not working !

thanks

flier268, 2023-07-27:
still an issue, so should be reopened