me-no-dev/ESPAsyncWebServer #650: Logout of authenticated page

Speedy2k, 2019-11-18:
Hi, i have a page that need login to have access to data, the problem is that i want to be able to log out before leaving so nobody else can get acess to the settings. but when i does this:

server.on("/logout", HTTP_GET, [](AsyncWebServerRequest * request){
    request->requestAuthentication();
    request->redirect("/");
  });

IT doesn't log out! if i return to this:

server.serveStatic("/advanced.html", SPIFFS, "/advanced.html").setCacheControl("max-age=600").setAuthentication((char*)loginUser.c_str(), (char*)loginPass.c_str());

It get right in without asking for password!?

me-no-dev, 2019-11-22:
this is because the auth header is still cached and used by the browser. You have to reset this header in the browser. I've done this... seems that what you wrote should have generally worked. Maybe instead of requestAuthentication then redirect, you should do requestAuthentication in the index? or maybe another code had to be sent to the browser 40something... but unless your browser stops using the headers, you will get auth every time...

sfranzyshen, 2020-02-03:
stale bot is not useful ... it degrades development ... please eliminate this feature from this repository ...

closed 2020-04-17 by stale[bot]