me-no-dev/ESPAsyncWebServer #653: Respond with multiple Strings from PROGMEM

ledsi, 2019-11-24:
So i'm not sure if i am doing it wrong but is there any way to respond with multiple Strings from PROGMEM? For example i will serve PROGMEM_header and PROGMEM_content and PROGMEM_footer so that my Stings get not to long or will hold redundant data.

ttlappalainen, 2019-12-02:
I have the same problem. Could AsyncProgmemResponse have contents as array and e.g. method addContent_P so the code would be:
AsyncProgmemResponse *response = request->beginResponse_P(200, "text/html", PROGMEM_header);
response ->addContent_P(PROGMEM_content);
response ->addContent_P(PROGMEM_footer);

ali80, 2019-12-29:
It would be great if you could also mix progmem and RAM contents in a response

ttlappalainen, 2019-12-29:
I made changes so that I can have e.g.

  ```
AsyncProgmemResponse *response = request->beginResponse_P(200,"text/html",(const uint8_t *)pageStart,pageStartLen);
  response->addContent((const uint8_t *)pageContent,pageContentLen);
  response->addContent((const uint8_t *)pageEnd,pageEndLen);
```
So all parts will be pointed to PROGMEM (=normal flash on ESP32).

I have not tried to do pull request - do not know would me-no-dev be interested.

Currently you can mix things by using printf. Unfortunately that builds page reserving same amount of ram. Really mixing without reserverving extra RAM would require building linked list of added contents.

If you think about just mixing e.g. sensor values to page content, I think it is better to have just static page on progmem and update just sensor values with javascript. Then you do not need to resend whole page every time.

ali80, 2019-12-29:
As you mentioned using printf is not possible if the response is too large, only solutions are SPIFFS or PROGMEM which makes the pages static, I'm currently using javascript to update the page, but if you have too many modifications it  can get really messy.
Tinkerman (Xose Pérez)  from [espurna](https://github.com/xoseperez/espurna)
also had a similar kind of problem and have an interesting blog post on this
[Embed your website in your ESP8266 firmware image](https://tinkerman.cat/post/embed-your-website-in-your-esp8266-firmware-image/)

ttlappalainen, 2020-02-27:
I made functionality for ESP32, which works as I described:
```
AsyncProgmemResponse *response = request->beginResponse_P(200,"text/html",(const uint8_t *)pageStart,pageStartLen);
response->addContent((const uint8_t *)pageContent,pageContentLen);
response->addContent((const uint8_t *)pageEnd,pageEndLen);
```
I use it on several different device, but have not published it, since I do not know is any other interested.

closed 2020-05-21 by stale[bot]