me-no-dev/ESPAsyncWebServer #657: Callback context on ESP32?

ashley-schaeffer, 2019-12-02:
I'm running ESPAsyncWebServer on ESP32, with AsyncTCP obviously.
I don't create any FreeRTOS tasks myself so I only have the Arduino loop() checking for some flags. Those flags are set from inside callbacks I passed to the ESPAsyncWebServer and AsyncTCP libraries.

However I only recently took a closer look at the source of the AsyncTCP library and realized a new task is created for calling the user supplied callbacks. Am I right in thinking that I should synchronize access between any variables accessed by both the loop() function and the callbacks (both in ESPAsyncWebServer and AsyncTCP) to prevent possible race conditions?

I'm asking because I haven't really seen this mentioned explicitly somewhere. I'm guessing it's only an "issue" on the ESP32?

BlueAndi, 2019-12-10:
Yes, you are right. The loop() is running in a task with prio 1 and the ESPAsyncWebserver is running in a separste task with prio 3. This means you have to consider concurreny access.

me21, 2019-12-24:
...unless variable access is atomic, that is, variable type is primitive (i.e. book, uint32) and both tasks use the same CPU core, if I understand correctly

mamama1, 2020-01-18:
hi,

this is kinda blowing my mind atm. could someone please be so kind and provide a simple example which demonstrates this issue? I might currently experience it but I'm not sure.

Thanks!

closed 2020-04-01 by stale[bot]