me-no-dev/ESPAsyncWebServer #659: Streaming a file using its handle

bot1131357, 2019-12-08:
I want to stream files from an SD card. 

Is there a method like `size_t WebServer::streamFile(T &file, const String& contentType)` in the arduino-esp32 library where I could specify the file handle (of type `File` using Sdfat 1.1.0 library)? Currently I make do with `void AsyncWebServerRequest::sendChunked(const String& contentType, AwsResponseFiller callback, AwsTemplateProcessor templateCallback)`, but there is a lot of manual wrangling of the data buffer. 

I'm not sure if this is the best approach. Is there a better method? Any suggestions or pointers would be much appreciated.

bot1131357, 2020-01-15:
Update: okay... not working so well as it crashes from time to time...

Voha888, 2020-02-13:
Hello. Does this code work? I used the beta version of SdFat and experimented with downloading large files. It’s best to use a streaming file `(WebServer::streamFile)` , but without handling or callback it has a lot of problems

bot1131357, 2020-02-14:
I am still looking for more elegant workarounds - as this code is unnecessarily complicated. I'd appreciate some pointers. Anyone?

Voha888, 2020-02-14:
I will probably use the last example from this topic.: https://github.com/me-no-dev/ESPAsyncWebServer/issues/124

Voha888, 2020-02-14:
@bot1131357 In your example, does the browser know about the file size?

bot1131357, 2020-02-16:
@Voha888 sorry didn't see your posts until now. 

Yes, the code does work - but it's something I've hacked up and honestly I don't think that's the best way to implement file download from SD card. 

Also, the example code does not mention file size, but you can change that with `addHeader()` method.

bot1131357, 2020-02-16:
Personally I would really like to see an ESPAsyncWebServer equivalent of WebServer::streamFile(), so that I can write this:
```
File anyfile = getFileHandle(); // SPIFFS, SDFAT, or any library that implements File interface
req->streamFile(anyfile , "application/zip");
```
Though at this point, I'm just not sure how to approach this...

Voha888, 2020-02-20:
I looked at the source of this function, it seemed to me that there it is not difficult to rewrite its work under SdFs (SdFat)

bot1131357, 2020-02-27:
> I looked at the source of this function, it seemed to me that there it is not difficult to rewrite its work under SdFs (SdFat)

Would you consider doing a pull request?

closed 2020-05-11 by stale[bot]