me-no-dev/ESPAsyncWebServer #663: Crash on request HTTP_GET when loading .html from SPIFFS

Dangio, 2019-12-16:
Hello,
I have a project to move a motor in forward (up) and backward (down) direction.
On the SPIFFS I have loaded an index.html and style.css page.
In my .INO file the page is loaded at start of program:

 Route for root / web page
`server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){request->send(SPIFFS, "/index.html", String(), false, processor);  });`

When I try to connect with the IP address the page is loaded. All perfect till here.

[image]

Clicking on UP or DOWN moves the motor with a PID controller to 660 pulses (+-) -> Changes a local variable User_Input. Clicking on the DOWN button executes a GET request /down

**Now I have 4 scenarios:**
_Code 1:_ 
`server.on("/down", HTTP_GET, [](AsyncWebServerRequest *request){
User_Input=-660; 
request->send(SPIFFS, "/index.html", String(), false, processor);  });`

After executing code 1 the esp8266 resets (rst cause 2, boot mode 3,6 ).

_Code 2:_ If I remove User_Input=-660;
`server.on("/down", HTTP_GET, [](AsyncWebServerRequest *request){ 
  request->send(**SPIFFS**, "/index.html", String(), false, processor);
  });`

After executing code 2  the browser goes back to index.html

_Code 3:_ If I remove User_Input=-660;
`server.on("/down", HTTP_GET, [](AsyncWebServerRequest *request){ 
User_Input=-660; 
  });`

After executing code 3 the motor moves to position -660, however the browser dont get a response back and does not load the main page index.html

**If I put the index.html and style.css file into .ino file (PROGMEM) the request from code1 works excellent**

It seems there is an issue changing a variable and sending a request to a .html file which is located in SPIFFS.

BlueAndi, 2019-12-17:
Try to call ```yield();``` after changing your local variable, then send the page from SPIFFS back.

Dangio, 2019-12-18:
Unfortunately it is not working.
Changed the code like below...

`server.on("/up", HTTP_GET, [](AsyncWebServerRequest *request){
    User_Input=660;   
    yield();
    request->send(SPIFFS, "/index.html", String(), false, processor);
  });`

...but get always a "ERR_CONNECTION_RESET" in my browser

BlueAndi, 2020-01-12:
Loading stuff from SPIFFS is very slow. Maybe you get a watchdog reset.
What does the crash dump says?

TurtleForGaming, 2020-01-16:
I have the same issue here my exception decoder log:
```
xception 0: Illegal instruction
PC: 0x4021102c: optimistic_yield(uint32_t) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 109
EXCVADDR: 0x00000000

Decoding stack results
0x40100263: sws_isr_15() at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/SoftwareSerial/SoftwareSerial.cpp line 48
0x4010044a: interrupt_handler at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring_digital.c line 136
0x401004d0: interrupt_handler at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring_digital.c line 162
0x40100410: interrupt_handler at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring_digital.c line 130
0x40100410: interrupt_handler at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring_digital.c line 130
0x40202e58: spiffs_object_find_object_index_header_by_name_v at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_nucleus.c line 1683
0x40202e49: spiffs_object_find_object_index_header_by_name_v at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_nucleus.c line 1678
0x40100dcd: __wrap_spi_flash_read at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_phy.c line 267
0x4020fbd9: EspClass::flashRead(unsigned int, unsigned int*, unsigned int) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 639
0x40211cba: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 78
0x40100dcd: __wrap_spi_flash_read at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_phy.c line 267
0x4020fbd9: EspClass::flashRead(unsigned int, unsigned int*, unsigned int) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 639
0x40211ca0: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 67
0x40202e2b: spiffs_object_find_object_index_header_by_name_v at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_nucleus.c line 1673
0x402063b9: spiffs_phys_rd at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_cache.c line 153
0x40100410: interrupt_handler at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring_digital.c line 130
0x40203400: spiffs_obj_lu_find_entry_visitor at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_nucleus.c line 166
0x40202db4: spiffs_object_find_object_index_header_by_name_v at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_nucleus.c line 1662
0x40204d6e: spiffs_object_find_object_index_header_by_name at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_nucleus.c line 1694
0x40100263: sws_isr_15() at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/SoftwareSerial/SoftwareSerial.cpp line 48
0x401004e1: interrupt_handler at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring_digital.c line 166
0x402021e5: SPIFFS_open at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_hydrogen.c line 231
0x40100b3d: realloc at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1504
0x40210567: String::changeBuffer(unsigned int) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 156
0x40211749: SPIFFSImpl::open(char const*, fs::OpenMode, fs::AccessMode) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_api.cpp line 35
0x40210567: String::changeBuffer(unsigned int) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 156
0x4020fd68: fs::FS::open(char const*, char const*) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/FS.cpp line 221
0x4020fd95: fs::FS::open(String const&, char const*) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/FS.cpp line 207
0x4020afaf: AsyncStaticWebHandler::_fileExists(AsyncWebServerRequest*, String const&) at /home/samuel/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp line 152
0x40210567: String::changeBuffer(unsigned int) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 156
0x4021078e: String::concat(char const*, unsigned int) at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 291
0x4020b138: AsyncStaticWebHandler::_getFile(AsyncWebServerRequest*) at /home/samuel/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp line 128
0x40210a00: String::startsWith(String const&) const at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 531
[... cut ...]
0x40210f38: esp_yield() at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40211016: __yield() at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 100
0x40208ed3: Linky::update(int) at /tmp/arduino_build_475517/sketch/linky.cpp line 103
0x4020768b: loop() at /home/samuel/LinkyLink/software/ESP8266_Linky/ESP8266_Linky.ino line 276
0x40210fe4: loop_wrapper() at /home/samuel/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125
```

BlueAndi, 2020-01-16:
@TurtleForGaming can you link your code?

closed 2020-03-30 by stale[bot]