me-no-dev/ESPAsyncWebServer #666: abort() was called at PC 0x40144873 on core 1

Aldrin-John-Olaer-Manalansan, 2019-12-23:
I am trying to stream an ESP32 CAM using ESPAsyncWebserver+Websocket because it gives the feature of using only "ONE Port" for a webserver and websocket. And it works as I see my stream for a couple of 2 seconds, then it crashes with a message of:

> abort() was called at PC 0x40144873 on core 1
> 
> Backtrace: 0x400911a4:0x3ffb1e10 0x400913d5:0x3ffb1e30 0x40144873:0x3ffb1e50 0x401448ba:0x3ffb1e70 0x401443d3:0x3ffb1e90 0x401444c2:0x3ffb1eb0 0x40144479:0x3ffb1ed0 0x400d367f:0x3ffb1ef0 0x400d3a35:0x3ffb1f10 0x400d43f4:0x3ffb1f40 0x400d157a:0x3ffb1f60 0x400de0e1:0x3ffb1fb0 0x4008d8ed:0x3ffb1fd0

The Decoded Backtrace said:

> Decoding stack results
> 0x400911a4: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
> 0x400913d5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
> 0x40144873: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
> 0x401448ba: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
> 0x401443d3: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
> 0x401444c2: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
> 0x40144479: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
> 0x400d367f: AsyncWebSocketMessageBuffer::AsyncWebSocketMessageBuffer(unsigned char*, unsigned int) at C:\Users\acer\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp line 149
> 0x400d3a35: AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at C:\Users\acer\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp line 1221
> 0x400d43f4: AsyncWebSocket::binaryAll(char const*, unsigned int) at C:\Users\acer\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp line 984
> 0x400d157a: loop() at C:\Users\acer\Desktop\CameraWebServer_aldrin/CameraWebServer_aldrin.ino line 323
> 0x400de0e1: loopTask(void*) at C:\Users\acer\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
> 0x4008d8ed: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

I do not know What I did wrong here but, I just want to stream the jpeg image to all websocket clients connected to the webserver,websocketserver:

```
//wireless connection
#include <ESPAsyncWebServer.h>

//day time handler
#include <NTPClient.h>
#include <WiFiUdp.h>

//esp32 cam important functions
#include <esp_camera.h>
#include "camera_pins.h"

#define WiFi_TryConnect_TimeOut 20000.0 // ilang  milliseconds maghihintay ang esp32 sa pag connect
#define builtin_serialkey "Qa!e6" // eto yung UNIQUE Serial Key ng Device nato
#define builtin_AP_SSID "IP Camera: Qa!e6" // eto yung SSID ng WiFi na gagawin ng device nato
#define builtin_AP_Password "Ba0!*sL8" // eto yung password ng WiFi na gagawin ng device nato

#define Port_WebServer 8000

#define standby_timer 2000

#define tunnel_webserver "ws://192.168.1.184:8000/ws"

#define flashlight 4
#define clientindicator_LED 12
#define lightdependentresistor 13
#define LDR_Sensitivity 2048

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org", 28800, 3600e3);

IPAddress staticip;
word customtimer = 0;
unsigned long timesincelastupdate = 0;

String ssid = "Manalansan";
String password = "Manalansan@123!";

/*
        <style> *{ margin:0; } img { display: absolute; margin: 0 !important; height: 100vh;  width: 100vw; } </style>

  "        <style>\n"
  "          img\n"
  "          {\n"
  "            display: block;\n"
[... 319 lines cut ...]
}

String getPublicIP()
{
  WiFiClient client;
  if (client.connect("api.ipify.org", 80))
  {
    client.println("GET / HTTP/1.0");
    client.println("Host: api.ipify.org");
    client.println();
  }
  else return String();

  while (!client.available()) {
    delay(500);
  }

  String line;

  while (client.available())
    line = client.readStringUntil('\n');

  return line;
}
```

closed 2020-03-06 by stale[bot]

athusrv, 2020-04-14:
Up!

I'm getting a similar issue! Did you find a solution @Aldrin-John-Olaer-Manalansan .
@me-no-dev would mind taking a look at this?

athusrv, 2020-04-14:
Okay! I found the problem. In my case, I was running out of memory, so when parsePacket function from WiFiUdp (https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiUdp.cpp#L45) was trying to allocate memory then it was crashing because I had no memory available. The problem was being caused because I am using TFT_eSPI library which contains a TFT_eSprite class that requires you to free the sprite once you push that sprite to the screen. It consumes a lot of memory and that's why it was causing issues for me. Hope it helps.

cgmckeever, 2020-08-06:
@athusrv 

A) How did you figure this out?
B) How did you fix it?
- Did you figure out a way to free the memory?

I have the same issue trying to stream an ESP32-CAM ..