me-no-dev/ESPAsyncWebServer #667: Basic Authentication still not working

philbowles, 2019-12-27:
I raised this issue in August [here](https://github.com/me-no-dev/ESPAsyncWebServer/issues/575). Another user said it was working for him - although he didn't say which version he was using. Maybe its a browser problem - but the code is clearly wrong, in my view.  I explain in detail why in the above link and the simple 2-character-change fix I proposed in August still works. For the record, I'm using Chrome Version 79.0.3945.88 (Official Build) (64-bit) on Windows 7 and Windows 10 - behaviour the same on both. Fix works the same on both.

Using the lib downloaded from the primary link on the homepage today, this code still fails, i.e. it refuses access even when correct credentials are entered: 
```cpp
#include <ESP8266WiFi.h>
#include <ESPAsyncWebServer.h>

const char* ssid = "XXXXXXXX";
const char* password = "XXXXXXXX";

AsyncWebServer server(80);

bool _webAuth(AsyncWebServerRequest *request){
    if(!request->authenticate("admin","admin",NULL,false)) {
      request->requestAuthentication(NULL,false); // force basic auth
      return false;
    }
    return true;
}

void setup(void) {
  Serial.begin(74880);
  WiFi.begin(ssid, password);
  WiFi.waitForConnectResult();
  server.on("/",HTTP_GET,[](AsyncWebServerRequest *request){ 
    if(!_webAuth(request)) return;
    request->send(200,"text/plain","hello admin");
  });  
  server.begin();
}
void loop(void) {}
```
 Making the following change in WebAuthentication.cpp fixes the error for reasons explained in #575 -base64_encode_expected_len returns the length of the *buffer* required to contain a given encoding, not the length of the encoded *hash* itself. base64_encode_expected_len includes the trailing '\0' and therefore is always 1 byte longer than the actual hash, thus the test always fails.

Omitting the test completely also works. The test is redundant anyway in my opinion since if the hashes are different lengths they will (correctly) fail anyway in line 52 - the only effect of the test is to fail a few nanoseconds earlier for mismatching hashes and delay authentication by the same amount for valid ones. The fact that omitting it (even when correct!) makes no measurable difference in either case tends to support that view.

[image]

Results in the correct behaviour:

[image]

philbowles, 2019-12-27:
As a result of my previous edit my current fix is as follows:
[image]

wilkerman265, 2020-02-01:
Yes, there is an error in the code. I used your fix and basic auth works now for me. Thank you.

closed 2020-04-15 by stale[bot]

DrA1ex, 2024-08-24:
Just encountered the same problem. The fix works, but not because `base64_encode_expected_len` returns an extra byte for `\0`. The real reason is that the ESP8266 version of this function calculate additional space for a trailing `\n`.

### ESP8266
```cpp
#define BASE64_CHARS_PER_LINE 72

#define base64_encode_expected_len_nonewlines(n) ((((4 * (n)) / 3) + 3) & ~3)
#define base64_encode_expected_len(n) \
       (base64_encode_expected_len_nonewlines(n) + ((n / ((BASE64_CHARS_PER_LINE * 3) / 4)) + 1))
```

### ESP32
```cpp
#define base64_encode_expected_len(n) ((((4 * n) / 3) + 3) & ~3)
```

So the correct way to fix the issue is:

```cpp
#ifdef ARDUINO_ARCH_ESP32
size_t encodedLen = base64_encode_expected_len(toencodeLen);
#else
size_t encodedLen = base64_encode_expected_len_nonewlines(toencodeLen);
#endif
    
if(strlen(hash) != encodedLen)
    return false;
```

DRSDavidSoft, 2024-10-01:
Just a heads up, the behavior has seemingly changed in mathieucarbou's release v3.3.5:
https://github.com/mathieucarbou/ESPAsyncWebServer

Since the change, it appears that this issue will be solved for good in the fork!