me-no-dev/ESPAsyncWebServer #670: server.removeHandler(new SPIFFSEditor(_username,_password));

g4m3r-87, 2020-01-02:
Hello.

I am using the following code:
```
server.addHandler(new SPIFFSEditor(_username,_password));
server.begin();
```
Everything works well. When calling the HTPP: //ESP_IP/edit page, a login and password are requested.

In the process, I want to change _username, _password. 

I tried to do (with a new username and password):
server.addHandler (new SPIFFSEditor (_username, _password));

However, this does not work.

Next, I tried to do this:
```
server.removeHandler(new SPIFFSEditor(_username,_password));
_username=new_username;
_password=new_password;
server.addHandler (new SPIFFSEditor (_username, _password));
```
That didn't work either.

Please help. How can I change the username and password to access the edit page in the process without restarting ESP8266?

BlueAndi, 2020-01-13:
You are calling removeHandler() with a new instance of SPIFFSEditor. This won't work. You have to store the added handler (you get it as return value of addHandler) and use this as argument for the removeHandler.

g4m3r-87, 2020-01-15:
Hello!

I tried like this:

```
   auto handler = server.addHandler(new SPIFFSEditor(_username,_password));
   server.removeHandler(&handler);

 _username=new_username;
 _password=new_password;
 auto handler = server.addHandler (new SPIFFSEditor (_username, _password));
```

does not work

BlueAndi, 2020-01-15:
Now you copied the handler. ;-)

This should work:
```
AsyncWebHandler* handler = &server.addHandler(new SPIFFSEditor(_username, _password));
server.removeHandler(handler);
```

closed 2020-02-08 by g4m3r-87