me-no-dev/ESPAsyncWebServer #674: Set header for serveStatic

montyx99, 2020-01-13:
Hello,

I'm working on a reactjs based esp web server. My problem is that, there isn't an acceptable sized multiple entry point solution for react, so I cannot set and host multiple react application from the SPIFFS. I try to find a workaround, and a cookie, or another header variable could solve my issue, that I can send with the serveStatic method, and the setFilter.

**The use Case**
I create a simple wifiManager alternative for my project to host my react app. The esp bootup workflow is the following:

1. Read the credentials from the EEPROM
2. Try to connect in station mode with the credentials
3. Serve the index.html of my react app
4. If step 2 is not working, change to AP mode
5. Serve the same index.html as in step 3, but with a header, which tells to my react app that, only the Wifi setup page can be shown, without any navigation
6. Store the given credentials in the EEPROM
7. Restart the wifi and connect to the given access point

Is there any way, I can solve this issue? If not is there any way to redirect on "/" but serve the static files for "/" in the same time? Because currently I got the "Too much redirection" message, when I try that. (I know, this is how needs to work it)

montyx99, 2020-01-13:
My current code:

`server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");`

What I'd like to use:

`server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html").setFilter(ON_STA_FILTER);
server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html").setHeader(getAPHeader()).setFilter(ON_AP_FILTER);`

proddy, 2020-01-13:
Worth checking out this awesome project: https://github.com/rjwats/esp8266-react. Uses ESPAsyncWebServer with multiple configurable end-points with both AP and a captive portal. Rick is busy porting the code from JavaScript to TypeScript too. You may get some ideas there for your question.

montyx99, 2020-01-14:
@probonopd thanks for the advice, but he doesn't solve this issue. He has only one entry point, and required jwt authentication from the beginning. This is a very awesome project by the way, just uses different mindset.

montyx99, 2020-02-13:
Is there any possibility to find a solution for my issue or is this fully out of scope?

atanisoft, 2020-02-13:
@montyx99 There are multiple ways to solve this:

1. Setup a default header that the webserver injects on every request, see [this](https://github.com/me-no-dev/ESPAsyncWebServer#adding-default-headers) reference.
2. Consider using the built-in template processor to inject a flag value to the react app rather than rely on a header being available, see [this](https://github.com/me-no-dev/ESPAsyncWebServer#template-processing) reference.

montyx99, 2020-02-13:
Thank you very much for the quick answer. I'm new in IoT coding, so I always thinking in a normal backend way, so these solutions avoid my thoughts. Let me check these, and I'm going to close the issue.

montyx99, 2020-02-13:
The second way won't work in my case. I have a separated build pipeline for my React app, so I cannot use the built-in template processor. My react app provided as static content after build.

atanisoft, 2020-02-13:
The templates run on the static content (your index.html file) which would contain a templated parameter which gets injected. So it could possibly still work, it would still be worthwhile to explore.

montyx99, 2020-02-20:
@atanisoft it's worth a time to investigate. Thanks for the idea.
During that I've investigated the default header solution that you've mentioned before. The DefaultHeaders one. My problem is that I try to use a header in case of station mode, but when it's failed, I change to AP mode. So in the same function I have to change the same header based on the Wifi mode. What is the proper way to do that? Honestly I'm a JS guy, I'm just learning the C++, and tried to find the solution in the code of this repo, but I cannot understand the following lines:
https://github.com/me-no-dev/ESPAsyncWebServer/blob/a84f16989aa472658df2ed13b0399c0cdf3d7ece/src/ESPAsyncWebServer.h#L458

```arduino
  DefaultHeaders(DefaultHeaders const &) = delete;
  DefaultHeaders &operator=(DefaultHeaders const &) = delete;
```
In line 459 I see that this is some kind of operator definition, but I cannot understand the equal sign on the end. How can I learn about that?

Thanks in advance!

atanisoft, 2020-02-20:
@montyx99 I'm not certain what that notation is myself. But I wouldn't worry too much about that syntax.

I would suggest only adding the header value *AFTER* you have connected or switched to softap. You might be able to add the same header a second time and it might replace the first but it will take up a small bit of more memory.

montyx99, 2020-02-20:
> @montyx99 I'm not certain what that notation is myself. But I wouldn't worry too much about that syntax.
>
> I would suggest only adding the header value _AFTER_ you have connected or switched to [...]

Thanks a lot. Is this any way to reinitialize the default headers instance?

atanisoft, 2020-02-20:
I don't think there is a method for that available today.

montyx99, 2020-02-20:
Thank you. Maybe I'm going to check it, and if it works I'll create a pull request.

closed 2020-05-06 by stale[bot]

boomsya, 2024-01-10:
the same question.
source code:
`server.serveStatic("/js/", LittleFS, "/js/").setCacheControl("max-age=31536000");`
give header to response: `Cache-Control: max-age=31536000`
but I need to setup additional header like: `expires: Thu, 31 Dec 2037 23:55:55 GMT`
This is needed for browser not asking static files again and again after refreshing/reopening the page.

somebody have ideas?