me-no-dev/ESPAsyncWebServer #677: POST request becoming multiple callbacks.

WattsC-90, 2020-01-15:
I am trying to POST from my website back to the ESP8266 I am using, this has a POST url/endpoint on it "/changeWifi". However, if I print out the "data" from the body, it seems like the JSON object is received in multiple parts, it is being sent as one from the website, so I am not sure what I am doing wrong.

I am using the ArduinoJson library: https://github.com/bblanchon/ArduinoJson

the server endpoint:

```
server.on("/changeWifi", HTTP_POST, postTest, uploadTest, saveWifiSettings);
```
the callback methods:

```
void postTest(AsyncWebServerRequest* request) { Serial.println("post"); }
void uploadTest(AsyncWebServerRequest* request, const String& filename, size_t index, uint8_t* data, size_t len, bool final) {
	Serial.println("upload");
};
void saveWifiSettings(AsyncWebServerRequest* request, uint8_t* data, size_t len, size_t index, size_t total) {
	Serial.println("post..");
	DynamicJsonDocument doc(200);
	Serial.println((char*)data);
	DeserializationError err = deserializeJson(doc, (char*)data);
	String ssid = doc["SSID"];
	String pw = doc["password"];
	Serial.println(ssid);
	Serial.println(pw);
	//if (err) {
	//	request->send(500, "text/plain",err.c_str());
	//	return;
	//}
	request->send(200, "application/json", "{}");
};
```

based on all the `println`'s, the serial terminal shows the following:
[image]

This shows that the SSID is being split over two callbacks: `{"SSID":"` and then `unknown","password":""}`

The JQuery (just in case someone is curious):
```
function sendData(data, url) {
    $.ajax({
        url: url,
        type: "POST",
        contentType: "application/json",
        data: JSON.stringify(data),
        dataType: "json"
    });
}
```
I have checked using the console in chrome, it seems to transmit OK, I am not sure what is going wrong, the full headers that chrome is applying are here:
[image]

Any help is much appreciated, I am a little out of my depth! the odd thing is that using POSTman it works and prints the SSID and password correctly!

willfaulds, 2020-03-21:
+1 Did you resolve you're issue?

WattsC-90, 2020-03-22:
@willfaulds I havent tried since, youre the first person to comment so the feature sat still, moved onto other issues in the code! I shall try the issue you link too, thanks for the tip!

closed 2020-06-04 by stale[bot]