me-no-dev/ESPAsyncWebServer #678: supporting wild cards in requests

WattsC-90, 2020-01-15:
Is it possible for request patterns to support wildcards? the more files that I add to the project I feel it would be easier to support `/js/*.js` style request pattern and handle which file is actually requested using a `switch case` statement in the method instead.

It does occur to me while writing this, that it may cause extra overhead and may be recommended to do this another way (or to just have every single file listed!).

Any guidance is much appreciated

solopix, 2020-01-16:
You can use a single command like

    server.serveStatic("/favicon.ico", SPIFFS, "/favicon.ico").setCacheControl("max-age=31536000");
    server.serveStatic("/js/", SPIFFS, "/js/").setCacheControl("max-age=31536000");

or without cache

    server.serveStatic("/favicon.ico", SPIFFS, "/favicon.ico");
    server.serveStatic("/js/", SPIFFS, "/js/");

WattsC-90, 2020-01-16:
I have been looking at these as you have mentioned it. i dont understand how it works though, do the files get compiled into the application? Do i have to include the files some other way in the final application?

BlueAndi, 2020-01-22:
The files must be in the SPIFFS filesystem.
If you are using platformio: https://docs.platformio.org/en/latest/platforms/espressif32.html#uploading-files-to-file-system-spiffs

closed 2020-03-22 by WattsC-90