me-no-dev/ESPAsyncWebServer #679: Error compiling return-type false

valbuz, 2020-01-16:
by compiling it ends with the error:

`control reaches end of non-void function [-Werror=return-type]`

what does it mean?

thanks for help
pat

solopix, 2020-01-16:
let's see the whole function ...

valbuz, 2020-01-16:
thanks for reply.
yes, i look at it, but for me.....sorry.
i hope get help here.

Pablo2048, 2020-01-17:
Without seeing exactly the place where the error pops up we are unable to help more than generic response: you forget ```return <something>;``` somewhere in your code, or you have forget to change function return parameter to ```void```.

valbuz, 2020-01-17:
Hi.
the error is in this function:

`//-------------------------------------------------------------------------------
/*  
 *  ValidateNTP validates the incoming NTP data. 
 *  Stores the incoming data when correct.
 *  Returns false when something is wrong.
 */
//-------------------------------------------------------------------------------
bool ValidateNTP(AsyncWebServerRequest *request){
  int paramsNr = request->params();
  if(paramsNr == 3){
    
    AsyncWebParameter* p = request->getParam(0);
    NTPstring = p->value();

    p = request->getParam(1);
    String NTPData = p->value();
    gmtOffset_hour = NTPData.toFloat();

    p = request->getParam(2);
    NTPData = p->value();
    daylightOffset_hour = NTPData.toFloat();
  }
 else {
    return false;
  } 
}`

the return is given.

Pablo2048, 2020-01-17:
No, it is not given! Better way is to use ```bool retval = false;```, before the last ```}``` put ```return retval;```, remove ```else{...``` and put ```retval = true;``` after ```dayightOffset_hour = NTPData.toFloat();```.

valbuz, 2020-01-17:
now, only 1 error

`Error compiling for board ESP32 Dev Module.`

no link of the corresponding code...

i try another Notebook and install all again....here it work's compiling! 
no idea what happens here.....

thanks!

closed 2020-01-17 by valbuz