me-no-dev/ESPAsyncWebServer #680: ESP32 AsyncWebSocketClient->text(String) doesn't make it to the client browser

mamama1, 2020-01-18:
Hi

I have a fairly huge project in the works where I have recently started to port the whole thing from ESP8266 to ESP32 as I'm starting to run out of memory on the ESP8266.

On the ESP8266 everything works fine, but on the ESP32, most (but not all, oddly) WebSocket messages don't make it to the client browser (Chrome on Windows 10).

in the void onWsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len) callback I'm parsing the JSON request, putting a JSON request together, solely from RAM and then I'm writing it to the AsyncWebSocketClient * client via client->text(mystring);

It isn't even a huge amount of data, in fact this is it, what I'm trying to send:

> {"cmd":"getNodeIdList","ids":[1]}

In the opposite, this message gets delivered most of the time, but not always (also odd):

> {"cmd":"getwifi","ap":false,"ssidap":"hc5gateway","pskap":"hc5gateway","ssidsta":"myssid","psksta":"mypsk"}

again, with ESP8266, this works 100% reliably, I'm really out of ideas. According to wireshark, the expected messages don't make it to the PC. unfortunately I don't know if they ever leave the ESP32.

Wifi connection is stable, but icmp rtt is a bit high(?), about 30-100ms. with the esp8266 I get 1-2ms.

I'd happily post the entire code but I'm afraid nobody would like to chew himself through it. So I'm asking whether there are any obvious things I might look at or what portions of the code could be interesting to find the culprit.

I'm using the latest library versions and the latest ESP32 arduino core release. IDE: VSCode with platformio.

Thanks for any advice.

tlrobinson, 2020-01-20:
I think I'm running into the same problem (except I'm using `client->binary()`). The first message always seems to make it through, but about 50% of the time all subsequent messages don't. `server->binaryAll()` seems to always work.

tlrobinson, 2020-01-20:
I tracked down my problem to my code calling `client->ping()` *before* sending a non-ping frame via `client->binary()` (or presumably `client->text()`)

Sometimes it would just send a single ping, then the binary message, in which case everything would work fine, but sometimes it would send 2 (or more) pings before the binary message, in which case no messages after the first would be sent:

[image]

Removing the `client->ping()` from my `WS_EVT_CONNECT` handler code (before sending a binary message) seems to make it work correctly 100% of the time

mamama1, 2020-01-21:
i do not send any pings which i could remove, unfortunately...

BlueAndi, 2020-02-05:
Hmm ... I am using websockets as well, but can not confirm any problem right now.

mamama1, 2020-02-18:
Thanks! For now I have let go of the ESP32 and kept continuing with the ESP8266 but I might move back to ESP32 at some point and then I'll continue investigating on that.

solopix, 2020-02-18:
Maybe you have to use AsyncTCP instead of ESPAsyncTCP. Just update your libraries and in platformio.ini:

lib_ignore = ESPAsyncTCP

closed 2020-05-02 by stale[bot]

egrekov, 2020-06-18:
I confirm, if you exclude ping from the code, the error disappears and everything works.
`void onWsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len){
    if(type == WS_EVT_CONNECT){
        Serial.printf("ws[%s][%u] connect\n", server->url(), client->id());
        client->printf("Hello Client %u :)", client->id());
        // client->ping();`

zekageri, 2020-11-12:
i'm sending the esp's uptime and the RTC time to the web via websocket every second.
I can not exclude this from my project and the error comes no matter what.
After some time the esp restarts with bad  tail and heap corruption.

I tried to increase the queue size and i put the ws.cleanupClients(); to my loop if some client is stuck.
But the problem presist. :(