me-no-dev/ESPAsyncWebServer #690: Ticker in callback

deadbird99, 2020-01-28:
Hi.

I use ESPAsyncWebServer to, among other stuff, server a "reboot" page for an ESP32.

The callback code is the following:

```
void ConfigPortal::serveExit(AsyncWebServerRequest *request) {
    String page(HTML_HEAD);
    page.replace("{PAGETITLE}", "Main menu");
    
    page += F(HTML_STYLE);
    page += F(HTML_BEGIN_PAGE);

    page.replace("{ESPNAME}", getChipID());

    page += F(HTML_MESSAGE_RESETTING);
    page += F(HTML_END_PAGE);

    //The page contains a Javascript 5 seconds countdown before the ESP resets, thus the delay.
    Ticker reset;
    reset.attach(5, []() {
        Serial.println("restart");
#ifdef ESP32
        ESP.restart();
#else
        ESP.reset();
#endif  
    });
    
    request->send(200, "text/html", page);
}
```

The served page displays a Javascript 5 seconds countdown until the ESP (should) reset. But it's not resetting. The lambda code isn't called at all.

Am I doing something wrong? Or is there another way to do this?

Thanks a lot for your help.

Pablo2048, 2020-01-29:
Hi,
your `Ticker reset;` declaration is not static but has dynamic scope only in `serveExit` so when exitting `serveExit` method the `reset` object will be automatically destroyed. Try to use `static Ticker reset;` or global declaration.

deadbird99, 2020-01-29:
Silly me...C++ good habits a so far away...
Anyway, thanks a lot for the enlightenment!

closed 2020-01-29 by deadbird99